The-OpenROAD-Project/OpenLane #552: Post Simulation error in output reg

Khalique13, 2021-08-19:
### Description
A clear and concise description of what the bug is.

### Environment
* Operating System Information: `Ubuntu -20.4`
* Docker Version: `docker --version`
    * If you're using Podman, please also include this information.
* OpenLane Tag in use: `Gate level Synthesis ` in the main OpenLane directory.

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

### Expected behavior
A clear and concise description of what you expected to happen.

### Logs
```
Add any relevant logs here. Please do ensure they're enclosed by the triple-backticks.
```

Manarabdelaty, 2021-08-19:
@Khalique13 Check https://github.com/The-OpenROAD-Project/OpenLane/issues/518. You need to pass `-DFUNCTIONAL -DUNIT_DELAY=#1` to iverilog. Also, if you are simulating the netlist with power pins, you need to make sure that you are connecting `VPWR` and `VGND` pins of the top level netlist to `1'b1` and `1'b0` respectively in the testbench. Also, you will need to pass `-DUSE_POWER_PINS` to iverilog to use the powered cell views.

donn, 2021-08-20:
Duplicate of #518, locked.

closed 2021-08-20 by donn

Khalique13, 2021-08-20:
Here I am getting syntax error in sky130_fd_sc_hd.v at line 5363

donn, 2021-08-20:
Please continue the discussion in #518, Khalique.