The-OpenROAD-Project/OpenLane #553: [WARNING]: Failed to compare PDKS

vijayank88, 2021-08-19:
### Description
Just installed latest openlane version. I made #make full-pdk. If I do #make test got below message. Is that cause any issue? But passed make test passed successfully. Got print below message. 
========
[INFO]: Git Version: 2021.08.19_04.19.49
[WARNING]: Failed to compare PDKS
Malformed sky130A SOURCES file: openlane 40ada4699c2bbee45b273d5683849331e963488a did not match regex.
[WARNING]: OpenLane may not function properly.
================
### Environment
* Operating System Information: `uname -a`
* Linux 3.10.0-1160.36.2.el7.x86_64 #1 SMP Wed Jul 21 11:57:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
* Docker Version: `docker --version` Docker version 20.10.8, build 3967b7d
    * If you're using Podman, please also include this information.
* OpenLane Tag in use: `python3 ./get_tag.py` in the main OpenLane directory. 2021.08.19_04.19.49

### Reproduction Material
Tried after fresh OpenLane installation. got above warning message
#make test

### Expected behavior
No warning for PDK reading

### Logs
```
bash-4.2$ ./flow.tcl -design spm -tag 2ndtest
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Git Version: 2021.08.19_04.19.49
[WARNING]: Failed to compare PDKS
Malformed sky130A SOURCES file: openlane 40ada4699c2bbee45b273d5683849331e963488a did not match regex.
[WARNING]: OpenLane may not function properly.
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openLANE_flow/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/vijayan/OPEN_LANE_FLOW/Aug17_OPENLANE/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/vijayan/OPEN_LANE_FLOW/Aug17_OPENLANE/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: Current run directory is /openLANE_flow/designs/spm/runs/2ndtest
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/vijayan/OPEN_LANE_FLOW/Aug17_OPENLANE/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...

```

dralabeing, 2021-08-20:
I tested and get the exact error on this version

donn, 2021-08-30:
Can either of you share your `$PDK_ROOT/sky130A/SOURCES` file?

In my build, it looks like this:
```
-ne openlane 
f7fd774a762629bc56d31a94ff03138fbc2e176c
-ne skywater-pdk 
c094b6e83a4f9298e47f696ec5a7fd53535ec5eb
-ne open_pdks 
13e0611f6a27fd6e601f8fe207773233f4f3883f
```

Yet, it appears that your `SOURCES` file looks different:

`openlane 40ada4699c2bbee45b273d5683849331e963488a`

vijayank88, 2021-08-30:
@donn  
openlane f7fd774a762629bc56d31a94ff03138fbc2e176c
skywater-pdk c094b6e83a4f9298e47f696ec5a7fd53535ec5eb
open_pdks 13e0611f6a27fd6e601f8fe207773233f4f3883f

donn, 2021-08-30:
Gotcha. Weird, I don't know why this is different. I'm on it.

PR #564 (2021-08-30): Further Versioning Fixes
+ Proper OS detection in installing
+ Version added to installing (version no longer reports as N/A)
~ Fix #553
~ Reordered install steps to be more logical
~ get_tag now usable by other python files
commit 746546c0 (2021-08-30): Further Versioning Fixes (#564)

+ Proper OS detection in installing
+ Version added to installing (version no longer reports as N/A)
~ Fix #553
~ Reordered install steps to be more logical
~ get_tag now usable by other python files
closed 2021-08-30 by donn