The-OpenROAD-Project/OpenLane #558: Getting the Error

ksooryakrishna1, 2021-08-24:
I am not able to run my design and it is giving error. Attached the error log file along with this. 
Installed Openlane in the root and if I use ./flow.tcl -design spm or ./flow.tcl -design inverter, working fine.  Please help to solve this problem.

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/placement/10-opendp.log"
Last 10 lines:
child process exited abnormally

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/dvsd_adder_12bit -dn dvsd_adder_12bit -t first_run -o /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/final_summary_report.csv -m /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run"
Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/placement/19-opendp.log"
Last 10 lines:
child process exited abnormally

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/dvsd_adder_12bit -dn dvsd_adder_12bit -t first_run -o /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/final_summary_report.csv -m /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run"
Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/placement/28-opendp.log"
Last 10 lines:
child process exited abnormally

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/dvsd_adder_12bit -dn dvsd_adder_12bit -t first_run -o /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/final_summary_report.csv -m /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run"
Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/placement/37-opendp.log"
Last 10 lines:
child process exited abnormally

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/dvsd_adder_12bit -dn dvsd_adder_12bit -t first_run -o /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/final_summary_report.csv -m /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run"
Last 10 lines:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
[... 77 lines cut ...]
EOFError: EOF when reading a line
grep: /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory

during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_opendp.tcl |& tee >&@stdout /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/placement/91-opendp.log"
Last 10 lines:
child process exited abnormally

during executing: "openroad -python /openLANE_flow/report_generation_wrapper.py -d /openLANE_flow/designs/dvsd_adder_12bit -dn dvsd_adder_12bit -t first_run -o /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/reports/final_summary_ [... cut ...]
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/dvsd_adder_12bit/runs/first_run/logs/routing/fastroute.log: No such file or directory

vijayank88, 2021-08-24:
@ksooryakrishna1  its look like same design issue. Try modify config.tcl https://github.com/The-OpenROAD-Project/OpenLane/issues/550 what they used.

ksooryakrishna1, 2021-08-25:
I modified the contents of my config.tcl file and the following contents are added. Can you tell me  where should i add FP_CORE_UTIL 20  contents? 

# turn off clock
set ::env(CLOCK_TREE_SYNTH) 0
set ::env(CLOCK_PORT) ""

set ::env(PL_SKIP_INITIAL_PLACEMENT) 1
set ::env(PL_RANDOM_GLB_PLACEMENT) 0

set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 34.165 54.885"
set ::env(PL_TARGET_DENSITY) 0.75

set ::env(FP_HORIZONTAL_HALO) 6
set ::env(FP_VERTICAL_HALO) $::env(FP_HORIZONTAL_HALO)

vijayank88, 2021-08-25:
@ksooryakrishna1 
set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 34.165 54.885" is this required die area go ahead with this. 
Else comment above this line and use the variable FP_CORE_UTIL 20