The-OpenROAD-Project/OpenLane #576: Error: or_opendp.tcl, 29 can't read "::env(PL_DIAMOND_SEARCH_HEIGHT)": no such variable

madgo1991, 2021-09-02:
### Description

Getting following error :
Error: or_opendp.tcl, 29 can't read "::env(PL_DIAMOND_SEARCH_HEIGHT)": no such variable

In README.MD file it says this switch has been removed. But it seems it is still looking for this in the script.
Kindly help.

### Environment
* Operating System Information: `Ubuntu`
* Docker Version: Docker version 20.10.7, build 20.10.7-0ubuntu1~18.04.1
*  OpenLane  version : 0.15

vijayank88, 2021-09-02:
@madgo1991 visit https://github.com/The-OpenROAD-Project/OpenLane and follow instructions. Update to latest version will resolve your issues.

madgo1991, 2021-09-02:
With the latest release (2021.09.02_03.15.28) of OpenLANE I am getting following error :

[ERROR STA-0402] initialize_floorplan -tracks is not a known keyword or flag.
Error: or_floorplan.tcl, 90 STA-0402
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_floorplan.tcl |& tee >&@stdout /openLANE_flow/designs/picorv32_cwolf/runs/02-09_06-05/logs/floorplan/3-verilog2def.openroad.log"

madgo1991, 2021-09-02:
Why does it keep saying I am running v0.15 when I have pulled a different version which is the latest version ? Please see below for reference. 

/home/openLane/openlane/designs$ docker run -it -v /home/openLane/openlane/:/openLANE_flow -v /home/openLane/openlane/pdks/:/home/openLane/openlane/pdks -e PDK_ROOT=$PDK_ROOT efabless/openlane:2021.09.02_03.15.28
Unable to find image 'efabless/openlane:2021.09.02_03.15.28' locally
2021.09.02_03.15.28: Pulling from efabless/openlane
2d473b07cdd5: Already exists 
a5130161a390: Pull complete 
ed5bd0e327e8: Pull complete 
ea41992fb410: Pull complete 
3e1b815e49c4: Pull complete 
d113bd01251e: Pull complete 
19b81d589309: Pull complete 
9e06cf6416dd: Pull complete 
06a51402378e: Pull complete 
e03745428b58: Pull complete 
0ac232d7e03b: Pull complete 
3bf521828ee8: Pull complete 
50c94ab73bae: Pull complete 
a767be064fa7: Pull complete 
2faad29a357d: Pull complete 
81d46fa3101d: Pull complete 
19d46bbe5c05: Pull complete 
72644181d6e1: Pull complete 
fff1d1f2bd18: Pull complete 
Digest: sha256:e2663878cd0d0c33e1770e9fbd50fa73d9bfd11a7e10a5ff55b151fcaee73dbc
Status: Downloaded newer image for efabless/openlane:2021.09.02_03.15.28
[root@85a949039c52 openLANE_flow]# ./flow.tcl -design picorv32_cwolf            
[INFO]: 
	___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|     ||  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	|  O  ||   _/    _]|  |  || |___ |     ||  |  ||    _]
	|     ||  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: v0.15

vijayank88, 2021-09-02:
@madgo1991 try install in new directory path.. keep old version as it is.

madgo1991, 2021-09-07:
Please find two logs. One with this error and other with latest version of openlane. 
[picorv_cwolf_02092021.log.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7119088/picorv_cwolf_02092021.log.zip)
[picorv_cwolf_07092021.log](https://github.com/The-OpenROAD-Project/OpenLane/files/7119099/picorv_cwolf_07092021.log)

vijayank88, 2021-09-07:
@madgo1991  I mentioned clearly to create new directory path  and use git clone. But your log says same directory path old OL and new tag as well. If you don't have any design data do backup and trash openlane directory /home/openLane/openlane and install newly. latest version of scripts/openroad/or_floorplan.tcl don't have -tracks keyword. 
You merging into existing directory and still same error repeats

donn, 2021-12-12:
Staled out.

closed 2021-12-12 by donn