The-OpenROAD-Project/OpenLane #578: Incorrect diode perimeter

stevvik, 2021-09-02:
The SPICE view in sky130A PDK for sky130_fd_sc_hd stdcells library (sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice) contains diode description with incorrect perimeter value. This results in incorrect LVS in OpenLane, where this diode is used as antenna diode.

In sky130_fd_sc_hd.spice file:

.subckt sky130_fd_sc_hd__diode_2 DIODE VGND VNB VPB VPWR
D0 VNB DIODE sky130_fd_pr__diode_pw2nd_05v5 pj=5.36e+06u area=4.347e+11p
.ends

But in layout (sky130_fd_sc_hd.gds) the diode perimeter is 0.63*2+0.69*2=2.64.
So PJ parameter must be fixed to pj=2.64e+06u.

I fixed this in my local PDK copy and LVS passes correctly.

I haven't checked, but this error can be in other libraries as well.

donn, 2022-08-30:
@RTimothyEdwards @d-m-bailey Can you take a look at this? We're not facing any LVS issues atm but the report does appear to check out

RTimothyEdwards, 2022-08-30:
Good catch.  It appears to originate from an error in magic---the contacts are not being treated as part of the diode in the perimeter calculation, so the contact perimeter is added to the total perimeter (even though the area calculation is correct).  I'll have to fix the error in magic first, then patch the value in the PDK (except that it's going to be wrong for all diodes in all cell libraries. . . ugh).

RTimothyEdwards, 2022-08-30:
I found and fixed the error in magic.  Checking the build of open_pdks sky130 now.

donn, 2022-09-06:
Fixed in #1322.

closed 2022-09-06 by donn