The-OpenROAD-Project/OpenLane #579: Incorrect resistor description

stevvik, 2021-09-02:
The SPICE view in sky130A PDK for sky130_fd_sc_hd stdcells library (sky130A/libs.ref/sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice) contains resistors instantiation with incorrect SPISE syntax. This results in incorrect LVS in OpenLane.

In sky130_fd_sc_hd.spice file:

.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
R0 VGND LO VNB sky130_fd_pr__res_generic_po w=480000u l=45000u
R1 HI VPWR VNB sky130_fd_pr__res_generic_po w=480000u l=45000u
.ends

According to SPICE format resistor must have two ports followed by model name, but there are 3 here.
Therefore Netgen LVS treats VNB as model name and returns instances mismatch.

VNB port must be removed from resistors instantiations, like:

.subckt sky130_fd_sc_hd__conb_1 VGND VNB VPB VPWR HI LO
R0 VGND LO sky130_fd_pr__res_generic_po w=480000u l=45000u
R1 HI VPWR sky130_fd_pr__res_generic_po w=480000u l=45000u
.ends

The same resistors are correctly instantiated in other libraries, for example in sky130_fd_io/spice/sky130_fd_io.spice

Incorrect instantiations are in
sky130_fd_sc_hd/spice/sky130_fd_sc_hd.spice
and
 sky130_fd_sc_hvl/spice/sky130_fd_sc_hvl.spice

I fixed this in my local PDK copy and LVS passes correctly.

d-m-bailey, 2021-09-02:
This has been fixed in open_pdks.

https://github.com/RTimothyEdwards/open_pdks/commit/50161dc040edf388d7e92fc9b41cebb8c0b2dbbf

stevvik, 2021-09-02:
How can I get these changes? Should I update OpenLane and remake PDK?

d-m-bailey, 2021-09-02:
@stevvik I believe updating OpenLane and then `make pdk` should solve some of the issues.

You might want to take a look at https://github.com/google/skywater-pdk/issues/276 also.

I'm working on a device LVS flow, if you want to coordinate. There's a #be-verification slack channel you can use too.

donn, 2022-08-30:
Should be fixed by now, we've updated Open_PDKs many times.

closed 2022-08-30 by donn