The-OpenROAD-Project/OpenLane #585: Error during  diode_insertion can't find opendb python module

20Mhz, 2021-09-03:
Hi, 
This is needed by: ./scripts/write_powered_def.py
Not sure where is this is suppose to come from, didn't see it on OpenROAD repo. please advise.
Thanks!
Ronald

20Mhz, 2021-09-03:
I think is here now: src/odb/src/swig/python

It seems it has change its name as well....

vijayank88, 2021-09-04:
@20Mhz also mention which version of openlane you are using?  Install latest openlane version and try it. still error repeats share source file along with steps to reproduce the issue

donn, 2021-09-04:
We can’t help you if you just nuke the issue template.

20Mhz, 2021-09-04:
OpenLane f7fd774a762629bc56d31a94ff03138fbc2e176c
OpenROAD 527cd5d8e82308baa1a46265c1338e6c1757858f

It seems the swig module has changed name, but it also broken backwards compatibility.
Related to: https://github.com/The-OpenROAD-Project/OpenROAD/issues/1028

20Mhz, 2021-09-09:
https://gist.github.com/20Mhz/be34125a6747a559b4da7f073043ee9f

donn, 2021-09-09:
Ronald, that's not a workaround, that's a fix! If you can open a PR, I'd be happy to approve it.

donn, 2021-09-17:
Incorporated as part of #609, thanks

closed 2021-09-17 by donn