The-OpenROAD-Project/OpenLane #587: Design fail with GRT-0170 during resizer routing timing check

vijayank88, 2021-09-06:
### Description
I've eth_top design which is successfully generated GDSII with prev OL version. Failing with latest version  of the OL tool.

### Environment
* Operating System Information: `uname -a`: Ubuntu 20.04
* Docker Version: `docker --version`  Docker version 20.10.8, build 3967b7d
* OpenLane Tag in use: `python3 ./dependencies/get_tag.py` in the main OpenLane directory. 2021.09.03_03.49.05

### Reproduction Material
For issue design same #497 

### Expected behavior
Need to pass all stages of flow without fail.

### Logs
[INFO GRT-0101] Running extra iterations to remove overflow.
[ERROR GRT-0170] Invalid index for position (73, 101).
Error: or_resizer_routing_timing.tcl, 38 GRT-0170
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_resizer_routing_timing.tcl |& tee >&@stdout /openLANE_flow/designs/eth_top/runs/9thRun/logs/placement/14-resizer_timing.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/eth_top/runs/9thRun/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for eth_top/06-09_06-24 in 0h13m49s
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "openroad -python /openLANE_flow/generate_reports.py -d /openLANE_flow/designs/eth_top -dn eth_top -t 9thRun -o /openLANE_flow/designs/eth_top/runs/9thRun/reports/final_summary_report.csv -m /openLANE_flow/designs/eth_top/runs/9thRun/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/eth_top/runs/9thRun/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/eth_top/runs/9thRun"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
sed: can't read /openLANE_flow/designs/eth_top/runs/9thRun/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/eth_top/runs/9thRun/logs/routing/fastroute.log: No such file or directory
Traceback (most recent call last):
  File "/openLANE_flow/generate_reports.py", line 61, in <module>
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openLANE_flow/scripts/report/report.py", line 133, in get_report
    self.formatted_report = self.format_report()
  File "/openLANE_flow/scripts/report/report.py", line 120, in format_report
    splited_report[dieareaIdx] = str(float(splited_report[dieareaIdx])/1000000)
ValueError: could not convert string to float: 'E404'

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/eth_top/runs/9thRun/error.log

    while executing
"try_catch $::env(OPENROAD_BIN) -python $::env(OPENLANE_ROOT)/generate_reports.py -d $::env(DESIGN_DIR)  -dn $::env(DESIGN_NAME)  -t $::env(RUN_TAG)  -..."
    (procedure "generate_final_summary_report" line 16)
    invoked from within
"generate_final_summary_report"
    (procedure "flow_fail" line 5)
    invoked from within
"flow_fail"
    (procedure "try_catch" line 25)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_resizer_routing_timing.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(resi..."
    (procedure "run_resizer_timing_routing" line 13)
    invoked from within
"run_resizer_timing_routing"
    (procedure "run_routing" line 9)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 10)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 354)
```

```

donn, 2021-09-06:
@kareefardi Your -1 to E404 change. Enjoy.

kareefardi, 2021-09-06:
I will work on handling properly `E404`, however it seems that something else is going on here. 
```
INFO GRT-0101] Running extra iterations to remove overflow.
[ERROR GRT-0170] Invalid index for position (73, 101).
Error: or_resizer_routing_timing.tcl, 38 GRT-0170
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_resizer_routing_timing.tcl |& tee >&@stdout /openLANE_flow/designs/eth_top/runs/9thRun/logs/placement/14-resizer_timing.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
```
I think there is an option to turn off resizer timing optimizations. @marwaneltoukhy will know more about it

marwaneltoukhy, 2021-09-06:
@vijayank88 You can turn off the resizer timing optimization by setting  ```set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) 0``` in your config.tcl, this should fix the problem for now. Will work on a fix ASAP.

vijayank88, 2021-09-09:
@marwaneltoukhy with the update you gave more then 30+Hrs still running at routing stage. Previous build of OL took only 5Hrs to complete routing stage.

vijayank88, 2021-09-23:
With OL version 2021.09.19_20.25.16 able to resolve the issues and completed flow successfully.

closed 2021-09-23 by vijayank88