The-OpenROAD-Project/OpenLane #590: libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib does not exist when trying to do make pdk

atorkmabrains, 2021-09-06:
### Description
Trying to do a basic installation for Openlane. I followed the procedure outlined in the Readme.md. But it seems that there is an issue in the make pdk installation.

### Environment
* Operating System Information: Ubuntu 20.04
* Docker Version: Docker version 20.10.8, build 3967b7d rootless
* OpenLane Tag in use: 2021.09.03_03.49.05

### Reproduction Material
* git clone <Openlane_git_https_link>
* make openlane
* make pdk
* make test

### Expected behavior
Make test fails

### Logs
```cd /home/atorkmabrains/mabrains/OpenLane && \
	docker run --rm -v /home/atorkmabrains/mabrains/OpenLane:/openLANE_flow -v /home/atorkmabrains/mabrains/OpenLane/pdks:/home/atorkmabrains/mabrains/OpenLane/pdks -e PDK_ROOT=/home/atorkmabrains/mabrains/OpenLane/pdks  -u 0 efabless/openlane:2021.09.03_03.49.05 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.09.03_03.49.05
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openLANE_flow/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/atorkmabrains/mabrains/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openLANE_flow/designs/spm/config.tcl
[INFO]: Current run directory is /openLANE_flow/designs/spm/runs/openlane_test
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
/home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.tech/openlane/sky130_fd_sc_hd/drc_exclude.cells
[ERROR]: during executing: "/openLANE_flow/scripts/libtrim.pl /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib /openLANE_flow/designs/spm/runs/openlane_test/tmp/trimmed.lib.exclude.list > /openLANE_flow/designs/spm/runs/openlane_test/tmp/trimmed.lib"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Couldn't open /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib at /openLANE_flow/scripts/libtrim.pl line 38.

[ERROR]: Please check /openLANE_flow/scripts/libtrim.pl  log file
[ERROR]: Dumping to /openLANE_flow/designs/spm/runs/openlane_test/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for spm/06-09_14-07 in 0h0m6s
[INFO]: Generating Final Summary Report...
[ERROR]: during executing: "openroad -python /openLANE_flow/generate_reports.py -d /openLANE_flow/designs/spm -dn spm -t openlane_test -o /openLANE_flow/designs/spm/runs/openlane_test/reports/final_summary_report.csv -m /openLANE_flow/designs/spm/runs/openlane_test/reports/manufacturability_report.rpt -rs /openLANE_flow/designs/spm/runs/openlane_test/reports/runtime_summary_report.rpt -r /openLANE_flow/designs/spm/runs/openlane_test"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
sed: can't read /openLANE_flow/designs/spm/runs/openlane_test/logs/routing/fastroute.log: No such file or directory
sed: can't read /openLANE_flow/designs/spm/runs/openlane_test/logs/routing/fastroute.log: No such file or directory
Traceback (most recent call last):
  File "/openLANE_flow/generate_reports.py", line 61, in <module>
    report = Report(design, tag, design_name, params, run_path).get_report()
  File "/openLANE_flow/scripts/report/report.py", line 133, in get_report
    self.formatted_report = self.format_report()
  File "/openLANE_flow/scripts/report/report.py", line 120, in format_report
    splited_report[dieareaIdx] = str(float(splited_report[dieareaIdx])/1000000)
ValueError: could not convert string to float: 'E404'

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/spm/runs/openlane_test/error.log

    while executing
"try_catch $::env(OPENROAD_BIN) -python $::env(OPENLANE_ROOT)/generate_reports.py -d $::env(DESIGN_DIR)  -dn $::env(DESIGN_NAME)  -t $::env(RUN_TAG)  -..."
    (procedure "generate_final_summary_report" line 16)
    invoked from within
"generate_final_summary_report"
    (procedure "flow_fail" line 5)
    invoked from within
"flow_fail"
    (procedure "try_catch" line 25)
    invoked from within
"try_catch $::env(SCRIPTS_DIR)/libtrim.pl $arg_values(-input) $arg_values(-output).exclude.list > $arg_values(-output)"
    (procedure "trim_lib" line 25)
    invoked from within
"trim_lib"
    (procedure "prep" line 192)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 14)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 354)
make: *** [Makefile:223: test] Error 1
```

Also you could find the make pdk log at:
https://gist.github.com/atorkmabrains/7a780a82d5350816b2ad1dbb58b997ee

stat /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A

gives:
```
File: /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A
  Size: 4096      	Blocks: 8          IO Block: 4096   directory
Device: fd01h/64769d	Inode: 55840680    Links: 5
Access: (0755/drwxr-xr-x)  Uid: ( 1003/atorkmabrains)   Gid: ( 1004/atorkmabrains)
Access: 2021-09-06 16:03:35.180682688 +0200
Modify: 2021-09-06 16:03:46.512802928 +0200
Change: 2021-09-06 16:03:46.512802928 +0200
 Birth: -
```

stat /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib

gives:
```
stat: cannot stat '/home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib': No such file or directory
```

donn, 2021-09-06:
@RTimothyEdwards The main issue is `stat: '$PDK_ROOT/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib': No such file or directory`

I'm not really sure why this is occurring. You can find the full build log in the gist.

RTimothyEdwards, 2021-09-06:
@donn :  Failure to define $PDK_ROOT?  Everything else about the path is correct.

RTimothyEdwards, 2021-09-06:
@donn :  Although the statement itself sort of suggests that $PDK_ROOT was never expanded.

donn, 2021-09-06:
His pdk root exists otherwise, `stat /home/atorkmabrains/mabrains/OpenLane/pdks/sky130A/`. I replaced it with $PDK_ROOT for brevity's sake.

atorkmabrains, 2021-09-06:
I didn't define PDK_ROOT. I presume it's defined to "Installation_path"/pdks

closed 2021-11-23 by donn