The-OpenROAD-Project/OpenLane #594: spm xtea regression test failing with latest OL version

vijayank88, 2021-09-08:
### Description
Below command failed with non-zero exit status 1.
$python3 run_designs.py --designs spm xtea md5 aes256 --tag test --threads 3 

### Environment
* Operating System Information: `uname -a` Linux 012 5.11.0-27-generic #29~20.04.1-Ubuntu SMP Wed Aug 11 15:58:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
* Docker Version: `docker --version` : Docker version 20.10.8, build 3967b7d
 
* OpenLane Tag in use: `python3 ./dependencies/get_tag.py` in the main OpenLane directory. 2021.09.03_03.49.05

### Reproduction Material
$ python3 run_designs.py --designs spm xtea md5 aes256 --tag test --threads 3

### Expected behavior
complete the flow without fail.

### Logs
```
[2021-09-08 16:53:34,956 -  INFO] spm config_regression_0 running
[2021-09-08 16:53:34,962 -  INFO] spm config_regression_1 running
[2021-09-08 16:53:34,966 -  INFO] spm config_regression_2 running
[2021-09-08 16:53:34,989 -  INFO] spm config_regression_3 running
[2021-09-08 16:53:34,999 -  INFO] xtea config_regression_0 running
[2021-09-08 16:53:35,630 - ERROR] spm config_regression_0 failed check /home/vijayan/OPENLANE_FLOW/Aug15_2021_OpenLane/OpenLane/./designs/spm/runs/config_regression_0/error.txt
Exception in thread Thread-1:
Traceback (most recent call last):
  File "run_designs.py", line 314, in run_design
    subprocess.check_output(command.split(), stderr=subprocess.PIPE)
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['./flow.tcl', '-design', 'spm', '-tag', 'config_regression_0', '-overwrite', '-disable_output', '-config_tag', 'config_regression_0', '-no_save']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "run_designs.py", line 325, in run_design
    with open(run_path + "error.txt", "w") as error_file:
FileNotFoundError: [Errno 2] No such file or directory: '/home/vijayan/OPENLANE_FLOW/Aug15_2021_OpenLane/OpenLane/./designs/spm/runs/config_regression_0/error.txt'
Exiting thread 0

```

kareefardi, 2021-09-08:
Hello. The log file mentions `config_regression_*` implying that you are using `--regression`. Can you share the regression file that you are using?

vijayank88, 2021-09-08:
@kareefardi  sorry given wrong re-produce command. update the issue. Refer above again. Thanks.
My intention is to run different configuration for a design with single run and choose best report out of that.

kareefardi, 2021-09-08:
It seems that `./scripts/config/regression.config` is outdated. https://github.com/The-OpenROAD-Project/OpenLane/blob/ca93273185eb1f5650ea14afdb8659c56b980386/scripts/config/regression.config#L15 The value of `STD_CELL_LIBRARY` should be `sky130_fd_sc_hd`.

donn, 2021-09-08:
PR merged. Closing this.

closed 2021-09-08 by donn

vijayank88, 2021-09-09:
@donn  @kareefardi  I have updated config.tcl and still same error repeats. Are you able to run above commands on your workspace?