The-OpenROAD-Project/OpenLane #595: Build PDK part of the CI encountered error when doing git pull in open_pdks

erictaur, 2021-09-08:
### Description
Unable to build PDK using the OpenLane CI. All necessary files/scripts are included in the GitHub Action run.
Encountered a `fatal: No names found, cannot describe anything.` error when attempted to do `git pull` but not sure what the issue was.

@msaligane

### Environment
* Operating System Information: `GitHub Actions: runs-on: ubuntu-latest`

### Reproduction Material
* Do `make pdk` as a step of a job in GitHub Actions

### Expected behavior
The GitHub action run should be able to clone, pull and checkout to a specific  commit.

### Logs
```
make[1]: Leaving directory '/home/runner/work/OpenFASOC/OpenFASOC/pdks/skywater-pdk'
git clone https://github.com/rtimothyedwards/open_pdks /home/runner/work/OpenFASOC/OpenFASOC/pdks/open_pdks
                                                                                             
Cloning into '/home/runner/work/OpenFASOC/OpenFASOC/pdks/open_pdks'...
cd /home/runner/work/OpenFASOC/OpenFASOC/pdks/open_pdks && \
	git checkout master && git pull && \
	git checkout -qf 81d28fd0423ca6bbf3f3bf00b7d407958cf8f0f3
Already on 'master'
Your branch is up to date with 'origin/master'.
Already up to date.
fatal: No names found, cannot describe anything.
Please input the version of OpenLane you'd like to pull: 
Could not resolve the version of OpenLane being used. This is a critical error.
fatal: No names found, cannot describe anything.
Please input the version of OpenLane you'd like to pull: 
Could not resolve the version of OpenLane being used. This is a critical error.
```

donn, 2021-09-08:
Hey, this is because the OpenLane clone you're doing as part of GitHub actions is missing a fetch depth.

When you're missing the fetch depth, tags do not get pulled, so it fails to resolve the OpenLane version you need.

See https://github.com/Cloud-V/sky130-builds/blob/3d888bbb926265f1744c12fc8f4bbea02bd32501/.github/workflows/flow.yml#L19 as an example of how you should be doing that.

erictaur, 2021-09-09:
Hello @donn 
I was actually trying to replicate something like this:
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/.github/workflows/openlane_ci.yml#L115

In other words, cloning the pdk was NOT explicitly listed as a step in GHA. Instead, it ran `make pdk`, which clones `open_pdks` as seen in the Makefile at the repo root folder.

`make pdk` location:
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/Makefile#L68

Part where the git commands were run;
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/Makefile#L116

Per your suggestion, would I have to modify the git commands in the Makefile?

donn, 2021-09-09:
It's not the PDK clone that's really the issue here, it's the OpenLane clone that is.

How are you cloning OpenLane, is my question?

erictaur, 2021-09-09:
Hello @donn 
Sorry for misunderstanding, my apologies!

The Makefile pulls a docker image 
```
openlane:
	docker pull $(IMAGE_NAME)
```

With the following variables set as:
```
CURRENT_TAG ?= $(shell python3 ./.github/scripts/dependencies/get_tag.py)
IMAGE_NAME ?= efabless/openlane:$(CURRENT_TAG)
```
And in `get_tag.py` where I assume is the function that grabs the latest tags:
```
def get_tag():
    tag = None

    try:
        tag = subprocess.check_output(["git", "describe", "--tags", "--abbrev=0"]).decode("utf8").strip()
    except Exception as e:
        pass

    if tag is None:
        try:
            tag = open("./resolved_version").read()
        except:
            print("Please input the version of OpenLane you'd like to pull: ", file=sys.stderr)
            try:
                tag = input()
                with open("./resolved_version", "w") as f:
                    f.write(tag)
            except EOFError:
                print("Could not resolve the version of OpenLane being used. This is a critical error.", file=sys.stderr)
                exit(-1)

    return tag
```

If `subprocess.check_output(["git", "describe", "--tags", "--abbrev=0"]).decode("utf8").strip()` in `get_tag.py` won't work, then what are the contents of `./resolved_version`? 
`./resolved_version` is in the .gitignore so I am not sure if I would spot it.

@msaligane

donn, 2021-09-10:
> The Makefile pulls a docker image

To be very clear, I meant how are you cloning the OpenLane repository. What is, or was, the git command you used? This is one of the very few reasons that might cause `git describe --tags --abbrev=0` to fail.

---

`./resolved_version` is a filesystem cache for those using OpenLane without git, where they'd have to type in the OpenLane version manually via teletype. However, given that you are trying to run OpenLane in a CI environment, this is not particularly relevant.

erictaur, 2021-09-15:
Hello @donn 
Sorry for the late reply,

Thanks for clarifying some stuff!

However, I am not completely sure I understand your question,

But I can provide 2 things that are relevant to OpenROAD:

(1)
The `tool_metadata.yml` file (https://github.com/idea-fasoc/OpenFASOC/blob/main/.github/scripts/tool_metadata.yml) is being used at some point when the Makefile is pulling the docker image.

And in the file `tool_metadata.yml`, the relevant portion looks something like:
```
- name: openroad_app
  repo: https://github.com/The-OpenROAD-Project/OpenROAD
  commit: fb8ae93b6c7a5eb0e6fac83360a8a48d76c41885
  build: |
    mkdir -p ./build
    cd ./build
    cmake -DCMAKE_INSTALL_PREFIX=$PREFIX/bin ..
    make -j$(nproc)
```
(2)
I actually did the following in the YAML for GHA (This step is before the `make pdk` mentioned last time)
This step is successful 
```
      - name: Install OpenROAD
        run:  |
          cd /home/runner/work
          git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
          cd OpenROAD
          sudo ./etc/DependencyInstaller.sh -dev
          sudo ./etc/Build.sh
```

Sorry if my response is not comprehensive. If so, probably because I am still confused what's going on.

Thanks in advance!!

donn, 2021-09-15:
Well, the confusion here stems from the fact there are two similarly named projects, Open**ROAD** and Open**Lane**. You're showing me an Open**ROAD** clone here. I need the Open**Lane** clone.

erictaur, 2021-09-16:
Hello @donn 

Though experimental, I followed the setup in https://github.com/The-OpenROAD-Project/OpenLane and it seemed to work. The only thing now is to explicitly provide the location of open_pdks to the python script we are currently running.

I'll see if I can get it to work.

Thanks @donn 

FYI @mehdi

donn, 2021-09-17:
Welp, best of luck.

closed 2021-09-17 by donn