The-OpenROAD-Project/OpenLane #596: Cannot generate pdn for my design

peijunh, 2021-09-08:
Hello, my design is stuck at 7-pdn when I using the flow.tcl of OpenLane.

My design includes 5 macros, one is the sky130_sram_4kbyte_1rw1r_32x1024_8, the other four are my own sensors. All sensors use VDD and VSS instead of vccd1 stuff.

I know the naming might cause issue, so I write my own pdn.tcl file, which is shown below. I also add a set ::env(PDN_CFG) $::env(DESIGN_DIR)/pdn.tcl in the config.tcl. But the openlane is giving me this error: [ERROR PSM-0020] Cannot find net VPWR in the design. Please provide a valid VDD/VSS net.
Error: or_pdn.tcl, 39 PSM-0020

># Power nets
>set ::power_nets "VDD"
>set ::ground_nets "VSS"

>set pdngen::global_connections {
>  VDD {
>    {inst_name .* pin_name VPWR}
>    {inst_name .* pin_name VPB}
>    {inst_name .* pin_name VDD}
>    {inst_name .* pin_name vccd1}
>  }
>  VSS {
>    {inst_name .* pin_name VGND}
>    {inst_name .* pin_name VNB}
>    {inst_name .* pin_name VSS}
>    {inst_name .* pin_name vssd1}
>  }
>}
>##===> Power grid strategy
># Ensure pitches and offsets will make the stripes fall on track
>pdngen::specify_grid stdcell {
>    name block

>    rails {
>        met1 {width 0.49 offset 0}
>    }
>    straps {
>        met4 {width 1.60 pitch 3820.0 offset 25}
>        met5 {width 1.60 pitch 56.0 offset 2}
>    }
>    connect {{met1 met4} {met4 met5} {met4_PIN_ver met5} {met4_PIN_hor met5} {met3_PIN_ver met5} {met3_PIN_hor met5} {met2_PIN_ver met5} {met2_PIN_hor met5} {met1_PIN_ver met5} {met1_PIN_hor met5} {met4 met5} {met4_PIN_ver met4} {met4_PIN_hor met4} {met3_PIN_ver met4} {met3_PIN_hor met4} {met2_PIN_ver met4} {met2_PIN_hor met4} {met1_PIN_ver met4} {met1_PIN_hor met4} {li1_PIN_ver met4} {li1_PIN_hor met4} {li1_PIN_ver met5} {li1_PIN_hor met5}}
>    obstructions {}
>    pins {met1 met2 met3 met4 met5}
>pdngen::specify_grid macro {
>    orient {R0 R180 MX MY}
>    power_pins "VDD vccd1"
>    ground_pins "VSS vssd1"
>    blockages "met1 met2 met3 met4"
>    #straps {
>    #    met5 {width 1.6 pitch 84.16 offset 6.52}
>    #}
>    connect {{met3_PIN_hor met5} {met4_PIN_ver met5} {met3 met5} {met4 met5}}
>}

After that, I try to us the common_pdn.tcl. so I eliminate the set ::env(PDN_CFG) $::env(DESIGN_DIR)/pdn.tcl in the config.tcl and add the following line in config.tcl. However, this is giving me another error: [ERROR PDN-9999] Unexpected error: key "vccd1" not known in dictionary
PDN-9999

>set ::env(DESIGN_IS_CORE) 1
>set ::env(FP_PDN_CORE_RING) 1
>set ::env(VDD_NETS) [list {vccd1} {vccd2} {vdda1} {vdda2}]
>set ::env(GND_NETS) [list {vssd1} {vssd2} {vssa1} {vssa2}]
>set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"

Does anyone have any idea?

msaligane, 2021-09-08:
This issue is linked to: #537 # #

donn, 2021-09-09:
Correct. I've pinned the relevant issue as it has a workaround.

I'll be closing this.

msaligane, 2021-09-10:
@donn I think @peijunh already tried that workaround since he create his own pdn.tcl? @peijunh do you mind commenting on this?

peijunh, 2021-09-10:
@donn Yes, I do have create my own pdn.tcl and specify the parameters. I also use
> set ::env(PDN_CFG) $::env(DESIGN_DIR)/pdn.tcl 
as I mentions in the problem description. I guess I need to reopen the question.

msaligane, 2021-09-10:
ok, @peijunh I reopened this issue but please add more details regarding this issue maybe?
@donn he created a pdn.tcl file as suggested in the other issue so it is not really the same.

peijunh, 2021-09-10:
@donn please tell me what detail I shall add so that I can clarify my question. Thanks!

donn, 2021-09-10:
Beyond this point, we'd have to ask for the design files.

peijunh, 2021-09-13:
Hello: below is the version and our OS is centos7. But I am using docker for all the flow. 
commit f7fd774a762629bc56d31a94ff03138fbc2e176c
Author: Donn <me@donn.website>
Date:   Sun Aug 22 20:11:08 2021 +0300

    Makefile fixes (#551)

    *Actually* fixes #495. Oops.

donn, 2021-09-13:
We’d need the design files.

peijunh, 2021-09-13:
[opentitan_soc_top.floorplan.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7157403/opentitan_soc_top.floorplan.zip)

I got it. This is the design file. Thanks

msaligane, 2021-10-02:
@donn and @peijunh Any follow-up on this? this has been sitting too long..

donn, 2021-10-04:
A fix has been merged, can you try again and report back?

donn, 2021-12-12:
Staled out.

closed 2021-12-12 by donn