The-OpenROAD-Project/OpenLane #602: CVC step fails if MAGIC_EXT_USE_GDS=1

stevvik, 2021-09-09:
I set MAGIC_EXT_USE_GDS=1 in configuration file to activate Device level LVS and got CVC step (34) failed because awk is not able to find results/magic/<design_name>.lef.spice and crashes.

Here is a part of flow_summary.log:
[INFO]: current step index: 34
[INFO]: Running CVC
[ERROR]: during executing: "awk { /Black-box entry subcircuit/ {  # remove black-box defintions
    while ( $1 != ".ends" ) {
        getline;
    }
    getline;
}
/^\*/ {  # remove comments
    next;
}
/^.ENDS .*/ {  # remove name from ends lines
    print $1;
    next;
}
 {
    print $0;
}} /proj/memcom/memcom_tools/OpenTools/PDK/sky130/sky130A_2021.08.03_05.51.29/sky130A/libs.ref/sky130_fd_sc_hd/cdl/sky130_fd_sc_hd.cdl /openLANE_flow/designs/ram_dp_bist/runs/full_run/results/magic/ram_dp_bist.lef.spice > /openLANE_flow/designs/ram_dp_bist/runs/full_run/results/cvc/ram_dp_bist.cdl"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
awk: cmd. line:15: fatal: cannot open file `/openLANE_flow/designs/ram_dp_bist/runs/full_run/results/magic/ram_dp_bist.lef.spice' for reading (No such file or directory)

[ERROR]: Please check awk  log file
[ERROR]: Dumping to /openLANE_flow/designs/ram_dp_bist/runs/full_run/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Flow failed for ram_dp_bist/09-09_08-05 in 2h33m32s

d-m-bailey, 2021-09-09:
@stevvik Thanks for reporting this! We're working on a flow for device level LVS at the moment. In the mean time, you should be able to disable CVC by setting RUN_CVC = 0 in your config file.

stevvik, 2021-09-09:
@d-m-bailey Are you sure it is a good way to disable CVC during device level LVS? Maybe it would be better to make CVC workable even if device level LVS is enabled.

donn, 2021-11-17:
@d-m-bailey Does https://github.com/The-OpenROAD-Project/OpenLane/pull/686 address this?

donn, 2022-08-31:
@d-m-bailey Any updates?

kareefardi, 2023-09-09:
The flow no longer supports CVC

closed 2023-09-09 by kareefardi

d-m-bailey, 2023-09-11:
Just to be clear. CVC has been moved from openlane flow to mpw_precheck.