The-OpenROAD-Project/OpenLane #617: How to change the core_area and die_area of the project.

peijunh, 2021-09-18:
Hello, I am running my design with OpenLane flow. I specify the core area and die area in my config.tcl as below
>set ::env(DESIGN_IS_CORE) "1"
>set ::env(DIE_AREA) "0 0 7000 7000"
>set ::env(CORE_AREA) "50 50 6950 6950"

However, the resulting .def file in floorplan keep giving me this: DIEAREA ( 0 0 ) ( 655195 665915 ) ;
When I open the design/mydesign/runs/18-09_01-38/config.tcl, I find that there are two sets of AREA. One set is at the top of this config.tcl, which is exactly what I write in my config.tcl, also shown above. The other set is around 300 lines lower, which give me this:
>set ::env(CORE_AREA) "5.52 10.88 649.52 652.8
>set ::env(DIE_AREA) "0.0 0.0 655.195 665.915

And I have the error in the figure, which I guess is because my area is too small. Is there anyway to specify my own die area and core area?

vijayank88, 2021-09-18:
@peijunh Use absolute keyword followed by your die size.  use below format. Refer floorplan readme for more options.

set ::env(FP_SIZING) absolute
set ::env(DESIGN_IS_CORE) "1"
set ::env(DIE_AREA) "0 0 7000 7000"
set ::env(CORE_AREA) "50 50 6950 6950"

donn, 2021-09-18:
As vijayank88 wrote.

closed 2021-09-18 by donn