The-OpenROAD-Project/OpenLane #618: routing problem for my design

peijunh, 2021-09-19:
Hello. I am using OpenLane to run my design. I use to way to determine the size of my design, one is to use the following config to set the size of it manually
>set ::env(FP_SIZING) absolute
>set ::env(DESIGN_IS_CORE) "1"
>set ::env(DIE_AREA) "0 0 1000 1000"
>set ::env(CORE_AREA) "50 50 950 950"
This give me this error:
[image]
[image]

I then comment the AIE_AREA, CORE_AREA and use the default setting, which set the FP_CORE_UTIL to 50. But the result give me this error: [ERROR GRT-0118] Routing congestion too high.

Then I set FP_CORE_UTIL to 30 to see if I can lower the congestion. But I get the error in the following figure:
[image]

So is there any suggestion on why they are happening? Thanks!

donn, 2021-09-19:
1. Issue template
2. Please link the design

peijunh, 2021-09-19:
I just link the design and update the template @donn . Please check if I need to do anything else

vijayank88, 2021-09-20:
@peijunh  You can turn off the resizer timing optimization by setting set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) 0 in your config.tcl and try once

peijunh, 2021-09-20:
@vijayank88 Thanks for your help. I set :env(GLB_RESIZER_TIMING_OPTIMIZATIONS) to 0. I try to set PL_TARGET_DENSITY as 0.7, 0.55(default), 0.4, 0.35, but they all give me routing congestion too high. So may I ask what else I should set?

peijunh, 2021-09-20:
Looks like setting the PL_TARGET_DENSITY to 0.2 will help to the congestion problem. But there is problem in the triton route
```
[INFO DRT-0076]   Complete 100 pins.
[INFO DRT-0076]   Complete 200 pins.
[INFO DRT-0076]   Complete 300 pins.
[INFO DRT-0076]   Complete 400 pins.
[ERROR DRT-0073] No ap for periph_switch/u_s1n_10/DP_OP_12J5_122_1926/U14/A.
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0073
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_droute.tcl |& tee >&@stdout /openLANE_flow/designs/opentitan_soc/runs/20-09_20-26/logs/routing/19-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: SIGABRT

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/opentitan_soc/runs/20-09_20-26/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for opentitan_soc_top/20-09_20-26 in 0h4m35s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: opentitan_soc_top
Run Directory: /openLANE_flow/designs/opentitan_soc/runs/20-09_20-26
Source not found.
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/opentitan_soc/runs/20-09_20-26/results/lvs/opentitan_soc_top.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openLANE_flow/designs/opentitan_soc/runs/20-09_20-26/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.

    while executing
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_droute.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(tritonRoute_log_file..."
    (procedure "detailed_routing_tritonroute" line 6)
    invoked from within
"detailed_routing_tritonroute"
    (procedure "detailed_routing" line 13)
    invoked from within
"detailed_routing"
    (procedure "run_routing" line 49)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 10)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        puts_info "Running interactively"
        if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 354)
```

donn, 2021-09-22:
We'll look into it.

donn, 2021-09-23:
Okay, you've only included a `def` file and a config script.
I need the actual RTL design files.

peijunh, 2021-09-23:
Hi, attached is the compressed file of my design, you can directly put it under openLANE_flow/design/ and run I think. I use the netlist that is flattened by yosys because there are too many source RTL.
[opentitan_soc.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7218862/opentitan_soc.tar.gz)

peijunh, 2021-09-28:
@donn Hello, may I ask what mistake I made in my design? Thanks.

donn, 2021-09-29:
Yeah I'm. Getting more or less the same errors. I'll have to try again after updating OpenROAD.

msaligane, 2021-10-02:
@donn Have you been able to try this with latest of OpenROAD?

donn, 2021-10-04:
Have you?

msaligane, 2021-10-04:
I am assuming OL is pointing to the latest OR? so @peijunh please try again I guess.

donn, 2021-10-04:
Yup.

peijunh, 2021-10-04:
I clone OpenLane and build it from scratch. My commit now is shown below.
```
commit 27752ec11a12afa3214f64401662683f081e2644
Author: Openlane Bot <83099986+openlane-bot@users.noreply.github.com>
Date:   Wed Sep 29 22:03:01 2021 +0200

    [BOT] Update openroad_app (#616)

    openroad_app -> 7f23e5f6e6b8d36e6a8aaed6ad074902ea27c4ce
```

I run the flow again and get the following error. 

```
[INFO DRT-0165] Start pin access.
[INFO DRT-0076]   Complete 100 pins.
[INFO DRT-0076]   Complete 200 pins.
[INFO DRT-0076]   Complete 300 pins.
[INFO DRT-0076]   Complete 400 pins.
[INFO DRT-0076]   Complete 500 pins.
[ERROR DRT-0073] No ap for periph_switch/u_s1n_10/DP_OP_12J5_122_1926/U11/A.
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0073
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_droute.tcl |& tee >&@stdout /openLANE_flow/designs/opentitan_soc/runs/04-10_16-42/logs/routing/23-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: SIGABRT

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/opentitan_soc/runs/04-10_16-42/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for opentitan_soc_top/04-10_16-42 in 0h10m36s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: opentitan_soc_top
Run Directory: /openLANE_flow/designs/opentitan_soc/runs/04-10_16-42
Source not found.
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/opentitan_soc/runs/04-10_16-42/results/lvs/opentitan_soc_top.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openLANE_flow/designs/opentitan_soc/runs/04-10_16-42/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.
```

I am not sure why there is such and error, so I go to OpenLane/docker_build/docker/openroad_app/Dockerfile and find that at line 118 the commit is not the newest.
```
117 ARG OPENROAD_APP_REPO=https://github.com/The-OpenROAD-Project/OpenROAD
118 ARG OPENROAD_APP_COMMIT=4d4d7205fd0292dbf3fae55fad9109b3f0bd5786
```
May I ask how to fix this issue?

msaligane, 2021-10-04:
Thanks @peijunh 
I think it is failing in detail route now. I think it says that there are no access pins for this pin: `periph_switch/u_s1n_10/DP_OP_12J5_122_1926/U11/A`

Can you check what instance is that? and check versus the LEF we have used in MPW-II ? Also, maybe share your test case.

I am tagging @maliberty (just fyi)

peijunh, 2021-10-04:
This is the design file

[opentitan_soc.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7279963/opentitan_soc.tar.gz)

donn, 2022-08-30:
Staled out- please file a new issue if the problem persists.

closed 2022-08-30 by donn