The-OpenROAD-Project/OpenLane #625: Make test failed

akashpatil98, 2021-09-27:
### Description
I'm trying to setup the OpenLane environment by following the [documentation](https://openlane.readthedocs.io/en/latest/#running-openlane). But in the quick start, last command, when I run `sudo make test` I'm getting an error. 

### Environment
* Operating System Information: `Ubuntu 18.04.6`
* Docker Version: `20.10.8, build 3967b7d`
* OpenLane Tag in use: `2021.08.23_04.04.57` 

### Reproduction Material
As I am trying to setup the environment for the first time, I followed the steps mentioned in the documentation.
```console
git clone https://github.com/efabless/openlane.git
cd openlane/
make openlane
make pdk
make test
```

### Logs
```console
sudo make pdk
.
.
Magic 8.3 revision 183 - Compiled on Thu Jul  1 10:06:54 UTC 2021.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130(vendor): scaleFactor=2, multiplier=2
Scaled tech values by 2 / 1 to match internal grid scaling
Loading sky130A Device Generator Menu ...
Using technology "sky130A", version 1.0.225-0-g6c05bc4
CIF input style is now "sky130(vendor)"
Annotating cells from CDL/SPICE
Writing all magic database files
Writing '(UNNAMED)'
Must specify name for cell (UNNAMED).
Done.
Error message output from magic:
Can't write file named '(UNNAMED)'
Migrating LEF files to layout.
Warning:  No LEF files found in /home/akash/openlane/pdks/open_pdks/sky130/sky130A/libs.ref/sky130_sram_macros/lef
make[2]: Leaving directory `/home/akash/openlane/pdks/open_pdks/sky130'
if test "x" != "x" ; then \
	echo "Building OSU 12-track-high standard cell library" ;\
	make osu-t12-a ;\
fi
if test "x" != "x" ; then \
	echo "Building OSU 15-track-high standard cell library" ;\
	make osu-t15-a ;\
fi
if test "x" != "x" ; then \
	echo "Building OSU 18-track-high standard cell library" ;\
	make osu-t18-a ;\
fi
make[1]: Leaving directory `/home/akash/openlane/pdks/open_pdks/sky130'
echo "Ended sky130A PDK staging on "`date` >> sky130A_make.log
echo "Starting SKY130 PDK migration on "`date` > sky130A_install.log
../common/staging_install.py -std_format \
	-source /home/akash/openlane/pdks/open_pdks/sky130/sky130A \
	-target /home/akash/openlane/pdks/sky130A \
	-variable PDKPATH \
	-link_from none 2>&1 | tee -a sky130A_install.log
Installing in target directory /home/akash/openlane/pdks/sky130A
Removing files from target
Copying staging files to target
Done.
Changing local path references from /home/akash/openlane/pdks/open_pdks/sky130/sky130A to /home/akash/openlane/pdks/sky130A
[... 139 lines cut ...]
[ERROR]: Flow Failed.

    while executing
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_pdn.tcl  |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(pdn_log_file_tag).log ..."
    (procedure "gen_pdn" line 8)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 94)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 40)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 354)
Makefile:210: recipe for target 'test' failed
make: *** [test] Error 1
akash@akash-lenovo:~/openlane$ 
```

vijayank88, 2021-09-27:
@akashpatil98 latest OL version is 2021.09.23_03.17.13. Try to update and check once again

akashpatil98, 2021-09-27:
@vijayank88 Thank you very much! It worked.

Kind regards,
Akash Patil

closed 2021-09-27 by akashpatil98

muscular-soreness-Zz, 2021-10-08:
@vijayank88 Thank you for pointing this change out. By the way, can you or someone else update the documentation?