The-OpenROAD-Project/OpenLane #626: Macro hardening failed at final routing

vijayank88, 2021-09-27:
### Description
I am trying to do macro hardening for riscv_core.
Same riscv_core I'm able to complete gds2 generation in 20min using OpenRoad-flow-scripts.
But in openlane flow got stopped at final routing stage after 65Hrs time.
logs shown at the end

### Environment
* Operating System Information: `uname -a` Ubuntu 20.04
* Docker Version: `docker --version` 20.10.8, build 3967b7d
 * OpenLane Tag in use: `python3 ./dependencies/get_tag.py` in the main OpenLane directory.  2021.09.23_03.17.13

### Reproduction Material
* Find the below link to download source files. 
https://skywater-pdk.slack.com/archives/D026ENCEDK3/p1632745930000400
* List the commands used to run the design. Inside Openlane directory
 $make mount
 $./flow.tcl -design riscv_core

### Expected behavior
Lesser run time and complete the flow without routing violations error.

### Logs
```
[INFO DRT-0195] Start 64th optimization iteration.
    Completing 10% with 2730 violations.
    elapsed time = 00:00:00, memory = 1852.12 (MB).
    Completing 20% with 2730 violations.
    elapsed time = 00:00:00, memory = 1852.12 (MB).
    Completing 30% with 2730 violations.
    elapsed time = 01:37:23, memory = 1853.10 (MB).
    Completing 40% with 2730 violations.
    elapsed time = 01:37:23, memory = 1853.10 (MB).
    Completing 50% with 2730 violations.
    elapsed time = 02:43:34, memory = 1857.18 (MB).
    Completing 60% with 2730 violations.
    elapsed time = 02:43:34, memory = 1857.18 (MB).
    Completing 70% with 2730 violations.
    elapsed time = 03:50:20, memory = 1903.84 (MB).
    Completing 80% with 2729 violations.
    elapsed time = 04:11:09, memory = 1881.82 (MB).
    Completing 90% with 2729 violations.
    elapsed time = 04:23:58, memory = 1881.82 (MB).
    Completing 100% with 2729 violations.
    elapsed time = 05:54:39, memory = 1904.20 (MB).
[INFO DRT-0199]   Number of violations = 2729.
[INFO DRT-0267] cpu time = 09:06:27, elapsed time = 05:54:40, memory = 1907.05 (MB), peak = 2180.31 (MB)
Total wire length = 1694629 um.
Total wire length on LAYER li1 = 5222 um.
Total wire length on LAYER met1 = 552779 um.
Total wire length on LAYER met2 = 670121 um.
Total wire length on LAYER met3 = 314029 um.
Total wire length on LAYER met4 = 152476 um.
Total wire length on LAYER met5 = 0 um.
Total number of vias = 255777.
Up-via summary (total 255777):.

-------------------------
 FR_MASTERSLICE         0
            li1     96314
           met1    123816
           met2     26633
           met3      9014
           met4         0
-------------------------
                   255777

[INFO DRT-0198] Complete detail routing.
Total wire length = 1694629 um.
Total wire length on LAYER li1 = 5222 um.
Total wire length on LAYER met1 = 552779 um.
Total wire length on LAYER met2 = 670121 um.
Total wire length on LAYER met3 = 314029 um.
Total wire length on LAYER met4 = 152476 um.
Total wire length on LAYER met5 = 0 um.
Total number of vias = 255777.
Up-via summary (total 255777):.

-------------------------
 FR_MASTERSLICE         0
            li1     96314
           met1    123816
           met2     26633
           met3      9014
           met4         0
-------------------------
                   255777

[INFO DRT-0267] cpu time = 106:10:41, elapsed time = 64:43:54, memory = 1907.05 (MB), peak = 2180.31 (MB)

[INFO DRT-0180] Post processing.
Saving to /openLANE_flow/designs/riscv_core/runs/MHSep24R3/results/routing/19-riscv_core.def
[ERROR]: There are violations in the design after detailed routing.
[ERROR]: Total Number of violations is 2729
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for riscv_core/24-09_16-00 in 64h55m19s

```

vijayank88, 2021-09-27:
[riscv_core.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7236267/riscv_core.zip)

vijayank88, 2021-09-30:
@donn with below update in config.tcl am able to complete the flow by 1Hrs18Min
set ::env(PL_TARGET_DENSITY) "0.35"
set ::env(GLB_RT_MAXLAYER) "5"

dralabeing, 2021-10-04:
@vijayank88 Can you share a few details of the flow commands in both ORFS and OL just to clarify? It looks like ORFS was able to complete without any changes to placement density and max route layer specification?

vijayank88, 2021-10-05:
@dralabeing 
By default ORFS and OLFS use 0.55-0.65 as placement density.
I've read some documents says if we have more register in our RTL we have to use placement density range 0.4-0.6 and moderate or no register we can use density range as 0.5-0.7.
riscv_core as more registers, so tried with 0.3 initially in ORFS and tool stopped and suggested to use 0.31 as placement density. 
So I've used following variable and able to complete the flow in both ORFS and OLFS.
set ::env(PL_TARGET_DENSITY) "0.35"

maliberty, 2021-10-05:
What documents say that?

vijayank88, 2021-10-05:
@maliberty  general physical design flow documents

dralabeing, 2021-10-21:
@vijayank88 This time it fails in detailed placement not in DRT as previously reported? Please share details of your problem and debugging flow to show root cause.

donn, 2021-11-10:
For timing violations, please follow Dr. Shalan's timing document: https://docs.google.com/document/d/13J1AY1zhzxur8vaFs3rRW9ZWX113rSDs63LezOOoXZ8/edit#heading=h.9y68197ebff7

This is not a bug with OpenLane. This is something that will need some experimentation to achieve closure.

closed 2021-11-10 by donn