The-OpenROAD-Project/OpenLane #629: design configuration should be self contained, any relative paths should be resolved relative the configuration.tcl/.json location

ahmednofal, 2021-09-28:
The config.tcl/config.json file contain variables that store certain paths, those paths should be resolved relative to the location of the configuration file itself not the location of the execution (from where flow.tcl was called)

So the configuration might include a path like this:

`../../../file.gds` this will be resolved during runtime in relation to where the flow.tcl script was called from (not where the flow.tcl actually resides) 
To overcome this shortcoming most of the tcl configuration scripts follow the awkward convention of grabbing the configuration file's absolute path and prepending it to the beginning of a path like this:

```
set script_dir [file dirname [file normalize [info script]]]
set ::env(MACRO_PLACEMENT_CFG) $script_dir/macro.cfg
```

This does not take into account the configuration using JSON format because JSON does not have the ability to reference the containing directory (or any execution specific values), it is simply raw data(which is what a configuration file should be).

donn, 2021-09-29:
there is an environment variable called DESIGN_DIR that OpenLane exposes for precisely these situations. maybe there should be a way to expose it in json

ahmednofal, 2021-10-04:
JSON is a static configuration file format, you can not use it to reference other preceding keys' values.

closed 2021-12-29 by donn