The-OpenROAD-Project/OpenLane #634: Issue with adding a new design

akashpatil98, 2021-10-02:
### Description
Hi, I'm new to the OpenLane flow. I was following the instructions given in the [documentation](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/designs/README.md) to use the openlane flow for a new design. I've installed OpenLane correctly and even viewed the generated gds files for `make test` to verify the installation.
But I'm not able to run it on a new design. Please find the console output below.

### Environment
* Operating System Information: `Linux akash-lenovo 5.4.0-86-generic #97~18.04.1-Ubuntu SMP Sat Sep 18 03:11:22 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
* Docker Version: `Docker version 20.10.8, build 3967b7d`
* OpenLane Tag in use: `2021.09.30_02.12.16`

### Issue (Console output)
```console
akash@akash-lenovo:~/OpenLane$ sudo make mount
cd /home/akash/OpenLane && \
	docker run -it --rm -v /home/akash/OpenLane:/openLANE_flow -v /home/akash/OpenLane/pdks:/home/akash/OpenLane/pdks -e PDK_ROOT=/home/akash/OpenLane/pdks  --user 0:0 efabless/openlane:2021.09.30_02.12.16
[root@39607478ff50 openLANE_flow]# ./flow.tcl -design my_design -init_design_config
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.09.30_02.12.16
[INFO]: Running non-interactively
[INFO]: Creating design src directory /openLANE_flow/designs/my_design/src
[INFO]: Populating /openLANE_flow/designs/my_design/config.tcl..
[INFO]: Finished populating:
/openLANE_flow/designs/my_design/config.tcl 
Please modify CLOCK_PORT, CLOCK_PERIOD and add your advanced settings to /openLANE_flow/designs/my_design/config.tcl
[SUCCESS]: Done...
[root@39607478ff50 openLANE_flow]# exit
exit
```
After this, I exited from the docker environment (make mount) to copy `my_design.v` file into `OpenLane/designs/my_design/src` folder. This is to show the file structure inside the new project after copying the verilog file into src folder.
```console
akash@akash-lenovo:~/OpenLane/designs/my_design$ tree -L 2
.
├── config.tcl
└── src
    └── my_design.v

1 directory, 2 files
akash@akash-lenovo:~/OpenLane/designs/my_design$ 
```
After this, I again do `sudo make mount` to start the docker environment. Then I proceed with the steps mentioned in the documentation. When I run `./flow.tcl -design my_design` I get an error. Please find the console output below:
```console
akash@akash-lenovo:~/OpenLane$ sudo make mount
cd /home/akash/OpenLane && \
	docker run -it --rm -v /home/akash/OpenLane:/openLANE_flow -v /home/akash/OpenLane/pdks:/home/akash/OpenLane/pdks -e PDK_ROOT=/home/akash/OpenLane/pdks  --user 0:0 efabless/openlane:2021.09.30_02.12.16
[root@ec7113bb0028 openLANE_flow]# ./flow.tcl -design my_design  
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.09.30_02.12.16
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openLANE_flow/designs/my_design/config.tcl
[INFO]: Sourcing Configurations from /openLANE_flow/designs/my_design/config.tcl
[INFO]: PDKs root directory: /home/akash/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/akash/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openLANE_flow/designs/my_design/config.tcl
[INFO]: Current run directory is /openLANE_flow/designs/my_design/runs/02-10_05-39
[... 78 lines cut ...]
[ERROR]: Flow Failed.

    while executing
"try_catch [get_yosys_bin]  -c $::env(SYNTH_SCRIPT)  -l [index_file $::env(yosys_log_file_tag).log 0]  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 34)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 354)
[root@ec7113bb0028 openLANE_flow]# 
```

I'm not sure why I get an error saying no such module `my_design` exists.

Kind regards,
Akash

vijayank88, 2021-10-04:
@akashpatil98 
Understand the error: ERROR: No such module: my_design
Your given RTL not having modue name my_design. Try to use your module and design name same in Open lane flow.

akashpatil98, 2021-10-04:
Hi, @vijayank88 Thank you!
It is working perfectly after changing the module name in the Verilog file. 

Kind regards,
Akash

closed 2021-10-04 by akashpatil98