The-OpenROAD-Project/OpenLane #636: OpenSTA hang with tristate

mattvenn, 2021-10-06:
### Description

This bug was found by @anall. This reduced test case shows the issue:

https://github.com/anall/bug-openlane_hang/tree/reduced

OpenLane hangs at step 12 trying to print clock skew report.

### Environment

* Operating System Information: `uname -a`
Linux matt-desktop 5.11.0-37-generic #41~20.04.2-Ubuntu SMP Fri Sep 24 09:06:38 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

* Docker Version: `docker --version`

Docker version 20.10.7, build 20.10.7-0ubuntu1~20.04.2

### Reproduction Material
* Upload a tarball containing the relevant design - see repo linked above
* List the commands used to run the design:
* cd $OPENLANE_ROOT; make mount; ./flow.tcl -design bug-openlane_hang

### Expected behavior

The flow to finish.

### Logs

Last lines printed before hanging.

```
Startpoint: we (input port clocked by clk)
Endpoint: data (output port clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                          2.00    2.00 ^ input external delay
                  0.02    0.00    2.00 ^ we (in)
     1    0.00                           we (net)
                  0.02    0.00    2.00 ^ input3/A (sky130_fd_sc_hd__buf_1)
                  0.09    0.11    2.11 ^ input3/X (sky130_fd_sc_hd__buf_1)
     1    0.01                           net3 (net)
                  0.09    0.00    2.11 ^ _04_/A (sky130_fd_sc_hd__inv_2)
                  0.02    0.04    2.15 v _04_/Y (sky130_fd_sc_hd__inv_2)
     1    0.00                           _03_ (net)
                  0.02    0.00    2.15 v _09_/TE_B (sky130_fd_sc_hd__ebufn_1)
                  0.10    0.10    2.25 ^ _09_/Z (sky130_fd_sc_hd__ebufn_1)
     1    0.01                           data_c (net)
                  0.10    0.00    2.25 ^ _08_/A (sky130_fd_sc_hd__ebufn_4)
                  0.18    0.23    2.48 ^ _08_/Z (sky130_fd_sc_hd__ebufn_4)
     2    0.02                           data (net)
                  0.18    0.00    2.48 ^ data (inout)
     0    0.02                           data (net)
                                  2.48   data arrival time

                  0.00   10.00   10.00   clock clk (rise edge)
                          0.00   10.00   clock network delay (ideal)
                          0.00   10.00   clock reconvergence pessimism
                         -2.00    8.00   output external delay
                                  8.00   data required time
-----------------------------------------------------------------------------
                                  8.00   data required time
                                 -2.48   data arrival time
-----------------------------------------------------------------------------
                                  5.52   slack (MET)

min_max_report_end
clock_skew_report

```

mattvenn, 2021-10-06:
Here is the whole design: https://github.com/anall/sudoku-accelerator/tree/97e77a088f2bb9ecb7a40e3fa11b96c1ba332264
And this is the CTS log, where it shows it does identify the clock and makes a clock tree.

[12-cts.log](https://github.com/The-OpenROAD-Project/OpenLane/files/7294436/12-cts.log)

vijayank88, 2021-11-18:
> Here is the whole design: https://github.com/anall/sudoku-accelerator/tree/97e77a088f2bb9ecb7a40e3fa11b96c1ba332264 And this is the CTS log, where it shows it does identify the clock and makes a clock tree.
> 
> [12-cts.log](https://github.com/The-OpenROAD-Project/OpenLane/files/7294436/12-cts.log)

@mattvenn with suduku_puzzle design I'm able to complete the flow without any issue using openlane tag 2021.11.17_01.42.05.

kareefardi, 2023-06-06:
This is working now with latest OpenLane. Please reopen if there is still and issue.

closed 2023-06-06 by kareefardi