The-OpenROAD-Project/OpenLane #639: Latest Openroad compile fails with error: ‘class sta::dbSta’ has no member named ‘findLogicConstants’

dineshannayya, 2021-10-11:
### Description
i Have pull the latest Openroad Project, Here is Git log details
<pre><font color="#C4A000">commit 92dd19146f4fb83f25387b05435a2ee770366afa (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: f45cc928b 2754f4fb9
Author: Matt Liberty &lt;mliberty@eng.ucsd.edu&gt;
Date:   Fri Oct 8 22:41:13 2021 -0700

    Merge pull request #1154 from The-OpenROAD-Project-staging/warnings
    
    fix more compiler warnings
</pre>

Compile fails with below error:

<pre>cd /home/dinesha/workarea/tools/source/OpenROAD_10012020/build/src/dbSta/src &amp;&amp; /usr/bin/c++  -DBOOST_ALL_NO_LIB -DSPDLOG_COMPILED_LIB -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/include -I/usr/include/tcl -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/dbSta/src/../include -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/sta/include/sta -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/sta -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/sta/include -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/include -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/utl/include -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/include/odb -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/src/def/def -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/src/def/defzlib -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/src/lef/lef -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/odb/src/lef/lefzlib -I/home/dinesha/workarea/tools/source/OpenROAD_10012020/src/gui/include  -O3 -DNDEBUG   -pthread -std=c++17 -Wno-cast-qual -o CMakeFiles/dbSta.dir/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx.o -c /home/dinesha/workarea/tools/source/OpenROAD_10012020/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx
<b>/home/dinesha/workarea/tools/source/OpenROAD_10012020/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx:</b> In function ‘<b>void find_logic_constants()</b>’:
<b>/home/dinesha/workarea/tools/source/OpenROAD_10012020/build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx:7102:8:</b> <font color="#CC0000"><b>error: </b></font>‘<b>class sta::dbSta</b>’ has no member named ‘<b>findLogicConstants</b>’
 7102 |   sta-&gt;<font color="#CC0000"><b>findLogicConstants</b></font>();
      |        <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~~</b></font>
make[2]: *** [src/dbSta/src/CMakeFiles/dbSta.dir/build.make:85: src/dbSta/src/CMakeFiles/dbSta.dir/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx.o] Error 1
make[2]: Leaving directory &apos;/home/dinesha/workarea/tools/source/OpenROAD_10012020/build&apos;
make[1]: *** [CMakeFiles/Makefile2:2637: src/dbSta/src/CMakeFiles/dbSta.dir/all] Error 2
make[1]: Leaving directory &apos;/home/dinesha/workarea/tools/source/OpenROAD_10012020/build&apos;
make: *** [Makefile:133: all] Error 2
</pre>

vijayank88, 2021-10-11:
@dineshannayya can you share your config.tcl and src file

vijayank88, 2021-10-11:
@dineshannayya close this  and post your GitHub issue  in OpenROAD forum :
https://github.com/The-OpenROAD-Project/OpenROAD

dineshannayya, 2021-10-11:
Yes, Issue resolved after pulling the git sub module

closed 2021-10-11 by dineshannayya