The-OpenROAD-Project/OpenLane #658: CTS:: run_resizer_timing stage over-write the Synthesis netlist

dineshannayya, 2021-10-21:
During Synthesis **Output** =>  results/synthesis/*.synthesis_optimized.v
CTS  :: **run_cts** stage Output =>  results/synthesis/*.synthesis_cts.v
CTS ::  "run_resizer_timing" **Output**=> results/synthesis/*.synthesis_optimized.v  <= This over-write the Previous Synthesis netlist

The issue look to be due to File: scripts/tcl_commands/placement.tcl
proc run_resizer_timing {args} {
...
...
**write_verilog $::env(yosys_result_file_tag)_optimized.v**
}

Due this issue, Re-Running Only CTS Step once again is not working.

vijayank88, 2021-10-21:
@dineshannayya Just modify as per your need this write_verilog $::env(yosys_result_file_tag)_optimized.v to each stage and read same tag in next stage of flow.
Current stage CTS mean update to: write_verilog $::env(cts_result_file_tag)_optimized.v
But make sure reading same tag in next stage, by default scripts look for write_verilog $::env(yosys_result_file_tag)_optimized.v

vijayank88, 2021-10-21:
@dineshannayya Understood your query. 
In automated flow during placement read_verilog points synthesis.v. Post placement optimization write_verilog saved in _optimized.v. This is your CURRENT_NETLIST
While run_cts step verilog file written in _cts.v and this is your current netlist. Post CTS optimization i.e run_resizer_timing write_verilog happening and overwritten _optimized.v again.
Refer logs/flow_summary.log for more details.

If you're running in interactive mode point  read_verilog to the right file path.
Hope its clear.

dineshannayya, 2021-10-21:
Yes vijayank, I was bit confused with your previous comment: "follow instructions and share tar ball here" I thought you are looking for example setup from my side,

vijayank88, 2021-10-21:
@dineshannayya  if your query resolved mean close this issue.