The-OpenROAD-Project/OpenLane #683: flow.tcl no longer handles multiple simulation libraries.

d-m-bailey, 2021-10-30:
### Description
An attempt to copy `LIB_SYNTH_COMPLETE` as a file fails because it may be a list of files. 
In `scripts/tcl_commands/all.tcl`
```
        # trim resizer library
        if { ! [info exists ::env(LIB_RESIZER_OPT) ] } {
            set ::env(LIB_RESIZER_OPT) $::env(TMP_DIR)/resizer.lib
            file copy -force $::env(LIB_SYNTH_COMPLETE) $::env(LIB_RESIZER_OPT)       <- Attempt to copy list of files
            if { $::env(STD_CELL_LIBRARY_OPT) != $::env(STD_CELL_LIBRARY) } {
                set opt_lib $::env(TMP_DIR)/resizer_optlib.lib
                file copy -force $::env(LIB_SYNTH_OPT) $opt_lib
                lappend $::env(LIB_RESIZER_OPT) $::env(LIB_SYNTH_OPT)
            }
        }
```
For reference, here is how `SYNTH_LIB_COMPLETE` is handled in `scripts/tcl_commands/synthesis.tcl`
```
        set ::env(LIB_SYNTH_COMPLETE_NO_PG) [list]
        foreach lib $::env(LIB_SYNTH_COMPLETE) {
                set fbasename [file rootname [file tail $lib]]
                convert_pg_pins $lib $::env(TMP_DIR)/$fbasename.no_pg.lib
                lappend ::env(LIB_SYNTH_COMPLETE_NO_PG) $::env(TMP_DIR)/$fbasename.no_pg.lib
        }
```
From slack https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1635575865070700

### Environment
The actual error occurred on someone's EWS. However, I can verify the problem in the following installation.
```
Python: v3.7.3
Kernel: Linux v4.19.0-17-cloud-amd64
Distribution: debian 10
Container Engine: docker v20.10.7
OpenLane Git Version: 2021.10.25_20.35.00

Alert: Click is not installed.
---
PDK Version Verification Status: FAILED

Failed to compare PDKS
Environment variable PDK_ROOT is not set.
Traceback (most recent call last):
  File "/home/kanobailey/pull-requests/openlane/dependencies/verify_versions.py", line 55, in verify_versions
    raise Exception("Environment variable PDK_ROOT is not set.")
Exception: Environment variable PDK_ROOT is not set.

Failed to compare PDKs: Failed to compare PDKs.---
Git Log (Last 5 Commits)

commit ff6eec246a18d007bf450e94f289a460425a2883
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Thu Oct 28 11:06:49 2021 -0500

    Unify the way we are running STA  (#681)
    
    * Fix genericfir timing vios
    
    * Run sta inside the step script

commit 8d686c081c2c9aefa16dbbd8ccf5bc8f4dcabc4b
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Wed Oct 27 08:12:35 2021 -0500

    Update openroad + add multi-corner timing analysis (#680)
    
    * Update OR App
    
    * Fix overflow iterations
    
    * Fix repair_tie_fanout command
    
    * Update design config
    
    * Add multi-corner analysis

commit 683151cc9e601c2f2d774fbcc53cf23060787a41
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Wed Oct 27 09:36:07 2021 -0300

    Make sure docker logs dir exists (#679)
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>
    
    [skip ci]

commit 354a25f7743fe68fa02f6337a922594ac41b7cee
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Tue Oct 26 09:57:47 2021 -0500

    Update resizer options (#678)
    
    * Update design configuration to address timing violations
    
    * update resizer scripts
    
    - added repair_tie_fanout command
    - added catch to the repair_timing commands to prevent exiting when the utilization limit is reached
    
    * Reset routing_cores to 2

commit c0b75f5f37717201f90396efe835e1978e32dbe6
Author: Anton Blanchard <anton@ozlabs.org>
Date:   Wed Oct 27 00:11:33 2021 +1100

    Fix typo in python exit call (#677)
    
    In commit 4e158a77ef7a ("Quit on timing violations at the typical
    corner (#659)"), os.EX_DATA should have been os.EX_DATAERR.

```

### Reproduction Material
Any design with the `sky130_fd_sc_hvl` libraries will probably give the same error.

### Expected behavior
Should be able to handle multiple simulation libraries, I assume.

### Logs
From the slack conversation.
```
###############################################
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: N/A
[INFO]: Running non-interactively
[INFO]: Using design configuration at /project/openlane/user_proj_example/config.tcl
[INFO]: Sourcing Configurations from /project/openlane/user_proj_example/config.tcl
[INFO]: PDKs root directory: /home/booshan/OpenEDA
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/booshan/OpenEDA/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hvl
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hvl
[INFO]: Sourcing Configurations from /project/openlane/user_proj_example/config.tcl
[WARNING]: Removing exisiting run /project/openlane/user_proj_example/runs/user_proj_example
[INFO]: Current run directory is /project/openlane/user_proj_example/runs/user_proj_example
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/booshan/OpenEDA/sky130A/libs.ref/sky130_fd_sc_hvl/techlef/sky130_fd_sc_hvl.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
mergeLef.py : Merging LEFs
sky130_fd_sc_hvl.lef: SITEs matched found: 0
sky130_fd_sc_hvl.lef: MACROs matched found: 69
mergeLef.py : Merging LEFs complete
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
error copying " /home/booshan/OpenEDA/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30.lib /home/booshan/OpenEDA/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30_lv1v80.lib ": no such file or directory
    while executing
"file copy -force $::env(LIB_SYNTH_COMPLETE) $::env(LIB_RESIZER_OPT)"
    (procedure "prep" line 202)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 14)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "/openlane/flow.tcl" line 356)
make: *** [Makefile:43: user_proj_example] Error 1
```

vijayank88, 2021-11-05:
@donn @d-m-bailey 
While executing 
Executing "/openLANE_flow/scripts/libtrim.pl { /home/vijayan/OPEN_LANE_FLOW/7c395e1_OPENLANE_Au15/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30.lib /home/vijayan/OPEN_LANE_FLOW/7c395e1_OPENLANE_Au15/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30_lv1v80.lib } /openLANE_flow/designs/spm/runs/scHVL/tmp/trimmed.lib.exclude.list > /openLANE_flow/designs/spm/runs/scHVL/tmp/trimmed.lib"

After merging different libraries into single lib as trimmed.lib. 
This trimmed.lib used in or_replace.tcl with read_lib $::env(LIB_SYNTH) and getting following error.
'''
[ERROR STA-0026] /openLANE_flow/designs/spm/runs/scHVL/tmp/trimmed.lib line 31779, syntax error, unexpected KEYWORD, expecting $end.
Error: or_replace.tcl, 51 STA-0026
'''
Is that looking for library separator keyword while merging different library?

How to reproduce the issue:
Edit $OPENLANE_FLOW/designs/spm/config.tcl and add 
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"
Run the flow and will get above error

dralabeing, 2021-11-10:
@vijayank88 Please test the version of the script or_replace.tcl to see what string format is accepted for the trimmed.lib and suggest changes to fiz.

vijayank88, 2021-11-11:
@dralabeing  @donn 
With openlane sky130_fd_sc_hd STD_CELL_LIBRARY by default LIB_SYNTH set to sky130_fd_sc_hd__tt_025C_1v80.lib. See doc reference:
```
configuration/README.md:| `LIB_SYNTH` | The library used for synthesis by yosys. <br> (Default: `$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hd__tt_025C_1v80.lib`)
```
If user changing STD_CELL_LIBRARY to sky130_fd_sc_hvl user has to select LIB_SYNTH for PVT corner by setting below as one example. Like this user can select any available PVT libraries.
```
set ::env(LIB_SYNTH) "$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hvl__tt_025C_3v30.lib"
```
User has to take care of this setting. Also this should be added to document for new user understanding.

dralabeing, 2021-11-11:
@donn can you role this fix for or_replace.tcl for the changes you plan to make or should @vijayank88 make a pr?

donn, 2021-11-11:
pr

vijayank88, 2021-11-12:
@donn  @dralabeing I've missed "$::env(PDK_ROOT)/$::env(PDK)/libs.tech/openlane/$::env(STD_CELL_LIBRARY)/config.tcl" is the file having LIB_SYNTH library to read.
sky130_fd_sc_hd/hs/ms/ls/hdll all point to single LIB_SYNTH file. Only sky130_fd_sc_hvl has two LIB_SYNTH file.
```
# Technology lib

set ::env(LIB_SYNTH) "\
	$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hvl__tt_025C_3v30.lib\
	$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hvl__tt_025C_3v30_lv1v80.lib\
	"
set ::env(LIB_FASTEST) "$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hvl__ff_n40C_5v50.lib"
set ::env(LIB_SLOWEST) "$::env(PDK_ROOT)/$::env(PDK)/libs.ref/$::env(STD_CELL_LIBRARY)/lib/sky130_fd_sc_hvl__ss_150C_1v65.lib"

set ::env(LIB_TYPICAL) $::env(LIB_SYNTH)

```

OL flow works fine for single LIB_SYNTH with file copy. If more then one library, it should merged into single library.

Below scripts should be updated accordingly.
```
        # trim synthesis library
        set ::env(LIB_SYNTH_COMPLETE) $::env(LIB_SYNTH)
        set ::env(LIB_SYNTH) $::env(TMP_DIR)/trimmed.lib
        trim_lib

        # trim resizer library 
        if { ! [info exists ::env(LIB_RESIZER_OPT) ] } {
            set ::env(LIB_RESIZER_OPT) $::env(TMP_DIR)/resizer.lib
            file copy -force $::env(LIB_SYNTH_COMPLETE) $::env(LIB_RESIZER_OPT)
            if { $::env(STD_CELL_LIBRARY_OPT) != $::env(STD_CELL_LIBRARY) } {
                set opt_lib $::env(TMP_DIR)/resizer_optlib.lib
                file copy -force $::env(LIB_SYNTH_OPT) $opt_lib
                lappend $::env(LIB_RESIZER_OPT) $::env(LIB_SYNTH_OPT)
            }
        } 
```
@donn do you have any idea on merging library?

dralabeing, 2021-11-12:
@vijayank88 thanks for the investigation. Best that an OL team member fixes this based on the above

vijayank88, 2021-11-17:
#714 still not fixes this issue. 
Error log:
```
[INFO ODB-0222] Reading LEF file: /openlane/designs/spm/runs/def_hvllib/tmp/merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 69 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/spm/runs/def_hvllib/tmp/merged_unpadded.lef
Error: sta.tcl, 38 cannot read file  /home/vijayan/OPENLANE_FLOW/Aug15_2021_OpenLane/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30.lib /home/vijayan/OPENLANE_FLOW/Aug15_2021_OpenLane/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hvl/lib/sky130_fd_sc_hvl__tt_025C_3v30_lv1v80.lib .
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/sta.tcl |& tee >&@stdout /openlane/designs/spm/runs/def_hvllib/logs/synthesis/1-opensta"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally
```

How to reproduce:
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"
set above variable in designs/spm/config.tcl and execute
$./flow.tcl -design spm
Will reproduce same error as above

Requirement: Successful GDS generation

dralabeing, 2021-11-24:
@vijayank88 Could you retest and see if the above problem is now fixed?

vijayank88, 2021-11-25:
@dralabeing PR not updated yet. seems work in  progress

dralabeing, 2022-01-27:
@vijayank88 Please test and respond

vijayank88, 2022-01-27:
@donn  @dralabeing 
Failed at synthesis ABC mapping itself.
```
17. Executing ABC pass (technology mapping using ABC).

17.1. Extracting gate netlist of module `\spm' to `/tmp/yosys-abc-18osvC/input.blif'..
Extracted 253 gates and 350 wires to a netlist network with 97 inputs and 128 outputs.

17.1.1. Executing ABC.
ERROR: Can't open ABC output file `/tmp/yosys-abc-18osvC/output.blif'.
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/spm/runs/spm_hvl2/logs/synthesis/1-synthesis.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: spm
Run Directory: /openlane/designs/spm/runs/spm_hvl2
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/spm/runs/spm_hvl2/logs/finishing/spm.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/spm/runs/spm_hvl2/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(SYNTH_BIN)  -c $::env(SYNTH_SCRIPT)  -l [index_file $::env(synthesis_logs)/synthesis.log]  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 32)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 54)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 387)
```

donn, 2022-01-27:
That's it, I'm done. If anyone wants this fixed they can package an example for me to test.

closed 2022-01-27 by donn

vijayank88, 2022-01-27:
@donn 
Test spm built-in design: https://github.com/The-OpenROAD-Project/OpenLane/tree/master/designs/spm
Update config.tcl as follows and flow should complete successfully.
```
# Design
set ::env(DESIGN_NAME) "spm"

set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]

set ::env(CLOCK_PERIOD) "10.000"
set ::env(CLOCK_PORT) "clk"

set ::env(FP_PIN_ORDER_CFG) $::env(DESIGN_DIR)/pin_order.cfg

set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hvl"

set filename ./designs/$::env(DESIGN_NAME)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}

```

donn, 2022-01-27:
Appears to be an isolated incident with the spm crashing abc- APU works fine.

vijayank88, 2022-01-28:
@donn  please refer #893 for spm fails with hvl library.