The-OpenROAD-Project/OpenLane #693: make test not sucessfull

PhilipCJ20, 2021-11-06:
$ git clone https://github.com/The-OpenROAD-Project/OpenLane.git
$ cd OpenLane/
$ make openlane
$ make full -pdk
$ make test
I have followed the steps provided in the githhub. I am using a tensor book (lambda with ubuntu 20, latest version). Also provided the absolute path to the pdks. Below i have attached the terminal screen showing openlane unsuccessful. Please help. I do not know what is the reason here.

cd /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane && \
	docker run --rm -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane:/openlane -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks:/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks -e PDK_ROOT=/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks   --user 1000:1000 efabless/openlane:2021.10.25_20.35.00 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.10.25_20.35.00-9-g01a6d1b
[WARNING]: 0 1
2 3
4 5
The version of openroad_app installed in the environment does not match the one required by the OpenLane flow scripts (installed: 7f23e5f6e6b8d36e6a8aaed6ad074902ea27c4ce, expected: b79f266fe41540eabc064bcaddfe19ed715ac5c2)
[WARNING]: OpenLane may not function properly.
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: Current run directory is /openlane/designs/spm/runs/openlane_test
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparation complete
[INFO]: Running Synthesis...
[INFO]: current step index: 1
[INFO]: Changing netlist from 0 to /openlane/designs/spm/runs/openlane_test/results/synthesis/spm.synthesis.v
[INFO]: Running Static Timing Analysis...
[INFO]: current step index: 2
[INFO]: Synthesis was successful
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: current step index: 3
[INFO]: Core area width: 87.86
[INFO]: Core area height: 87.04
[WARNING]: Current core area is too small for a power grid
[WARNING]: !!! THE POWER GRID WILL BE MINIMIZED. !!!
[INFO]: Final Vertical PDN Offset: 14.643333333333333
[INFO]: Final Horizontal PDN Offset: 14.506666666666668
[INFO]: Final Vertical PDN Pitch: 29.286666666666665
[INFO]: Final Horizontal PDN Pitch: 29.013333333333335
[INFO]: Changing layout from 0 to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def
[INFO]: Running IO Placement...
[INFO]: current step index: 4
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def
[INFO]: Running Tap/Decap Insertion...
[INFO]: current step index: 5
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def to /openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def
[INFO]: Power planning the following nets
[INFO]: Power: VPWR
[INFO]: Ground: VGND
[INFO]: Generating PDN...
[INFO]: current step index: 6
[INFO]: current step index: 7
[INFO]: PDN generation was successful.
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/6-pdn.def
[INFO]: Running Placement...
[INFO]: Running Global Placement...
[INFO]: current step index: 8
[INFO]: Global placement was successful
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/6-pdn.def to /openlane/designs/spm/runs/openlane_test/tmp/placement/8-replace.def
[INFO]: Running Resizer Design Optimizations...
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/or_resizer.tcl |& tee /dev/null /openlane/designs/spm/runs/openlane_test/logs/placement/8-resizer_design_optimization.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 5.0000 [current_design]
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 34 input buffers.
[INFO RSZ-0028] Inserted 1 output buffers.
[ERROR STA-0402] repair_design -slew_margin is not a known keyword or flag.
Error: or_resizer.tcl, 62 STA-0402
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openlane/designs/spm/runs/openlane_test/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: flow failed for spm/2021.11.06_00.40.57 in 0h0m12s
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: spm
Run Directory: /openlane/designs/spm/runs/openlane_test
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/spm/runs/openlane_test/results/lvs/spm.lvs_parsed.gds.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/spm/runs/openlane_test/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.

    while executing
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_resizer.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(resizer_log_file_ta..."
    (procedure "run_resizer_design" line 7)
    invoked from within
"run_resizer_design"
    (procedure "run_placement" line 24)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 356)
make: *** [Makefile:214: test] Error 1

d-m-bailey, 2021-11-06:
@PhilipCJ20 This appears to be caused by an older version of openroad being used. These kind of problems are usually automatically resolved in a day or two when programs are synced.

I manually updated openroad and `make test` ran to completion.

In `dependencies/tool_metadata.yml` change the openroad commit.
```
- name: openroad_app
  repo: https://github.com/The-OpenROAD-Project/OpenROAD
  #commit: b79f266fe41540eabc064bcaddfe19ed715ac5c2
  commit: 50f0e32411c12d71438481b8b353a03b45537baa
  build: |
    mkdir -p ./build
    cd ./build
    cmake -DCMAKE_INSTALL_PREFIX=$PREFIX/bin ..
    make clean
    make -j$(nproc)
  in_install: false
```
Then
```
cd docker_build
make build-openroad_app
make merge
cd ..
make test
```

PhilipCJ20, 2021-11-06:
@d-m-bailey , i followed the same commands but its still returning the error. Please have a look.
make test
cd /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane && \
	docker run --rm -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane:/openlane -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks:/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks -e PDK_ROOT=/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks   --user 1000:1000 efabless/openlane:2021.10.25_20.35.00 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.10.25_20.35.00-9-g01a6d1b
[WARNING]: 0 1
2 3
4 5
The version of openroad_app installed in the environment does not match the one required by the OpenLane flow scripts (installed: 7f23e5f6e6b8d36e6a8aaed6ad074902ea27c4ce, expected: 50f0e32411c12d71438481b8b353a03b45537baa)
[WARNING]: OpenLane may not function properly.
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[WARNING]: Removing exisiting run /openlane/designs/spm/runs/openlane_test
[INFO]: Current run directory is /openlane/designs/spm/runs/openlane_test
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparation complete
[INFO]: Running Synthesis...
[INFO]: current step index: 1
[INFO]: Changing netlist from 0 to /openlane/designs/spm/runs/openlane_test/results/synthesis/spm.synthesis.v
[INFO]: Running Static Timing Analysis...
[INFO]: current step index: 2
[INFO]: Synthesis was successful
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: current step index: 3
[INFO]: Core area width: 87.86
[INFO]: Core area height: 87.04
[WARNING]: Current core area is too small for a power grid
[WARNING]: !!! THE POWER GRID WILL BE MINIMIZED. !!!
[INFO]: Final Vertical PDN Offset: 14.643333333333333
[INFO]: Final Horizontal PDN Offset: 14.506666666666668
[INFO]: Final Vertical PDN Pitch: 29.286666666666665
[INFO]: Final Horizontal PDN Pitch: 29.013333333333335
[INFO]: Changing layout from 0 to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def
[INFO]: Running IO Placement...
[INFO]: current step index: 4
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def
[INFO]: Running Tap/Decap Insertion...
[INFO]: current step index: 5
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def to /openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def
[INFO]: Power planning the following nets
[INFO]: Power: VPWR
[INFO]: Ground: VGND
[INFO]: Generating PDN...
[INFO]: current step index: 6
[INFO]: current step index: 7
[... 38 lines cut ...]
No antenna report found.
[INFO]: check full report here: /openlane/designs/spm/runs/openlane_test/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow Failed.

    while executing
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_resizer.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(resizer_log_file_ta..."
    (procedure "run_resizer_design" line 7)
    invoked from within
"run_resizer_design"
    (procedure "run_placement" line 24)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 356)
make: *** [Makefile:214: test] Error 1

PhilipCJ20, 2021-11-06:
@d-m-bailey 
Thanks a lot Bailey. it worked with the new change you mentioned. Please have a look into the terminal.

make test
cd /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane && \
	docker run --rm -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane:/openlane -v /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks:/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks -e PDK_ROOT=/home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks   --user 1000:1000 efabless/openlane:resizer-fix sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
[INFO]: 
	 ___   ____   ___  ____   _       ____  ____     ___
	/   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
	|   | |  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
	| O | |   _/    _]|  |  || |___ |     ||  |  ||    _]
	|   | |  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
	\___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: 2021.10.25_20.35.00-9-g01a6d1b
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openlane/designs/spm/config.tcl
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks
[INFO]: PDK: sky130A
[INFO]: Setting PDKPATH to /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library is set to: sky130_fd_sc_hd
[INFO]: Sourcing Configurations from /openlane/designs/spm/config.tcl
[WARNING]: Removing exisiting run /openlane/designs/spm/runs/openlane_test
[INFO]: Current run directory is /openlane/designs/spm/runs/openlane_test
[INFO]: Preparing LEF Files
[INFO]: Extracting the number of available metal layers from /home/philip/Philip/Studies/VLSI_Open_source_tools/vsdflow/work/tools/OpenLane/pdks/sky130A/libs.ref/sky130_fd_sc_hd/techlef/sky130_fd_sc_hd.tlef
[INFO]: The number of available metal layers is 6
[INFO]: The available metal layers are li1 met1 met2 met3 met4 met5
[INFO]: Merging LEF Files...
[INFO]: Trimming Liberty...
[INFO]: Generating Exclude List...
[INFO]: Generating Exclude List...
[INFO]: Creating ::env(DONT_USE_CELLS)...
[INFO]: Storing configs into config.tcl ...
[INFO]: Preparation complete
[INFO]: Running Synthesis...
[INFO]: current step index: 1
[INFO]: Changing netlist from 0 to /openlane/designs/spm/runs/openlane_test/results/synthesis/spm.synthesis.v
[INFO]: Running Static Timing Analysis...
[INFO]: current step index: 2
[INFO]: Synthesis was successful
[INFO]: Running Floorplanning...
[INFO]: Running Initial Floorplanning...
[INFO]: current step index: 3
[INFO]: Core area width: 87.86
[INFO]: Core area height: 87.04
[WARNING]: Current core area is too small for a power grid
[WARNING]: !!! THE POWER GRID WILL BE MINIMIZED. !!!
[INFO]: Final Vertical PDN Offset: 14.643333333333333
[INFO]: Final Horizontal PDN Offset: 14.506666666666668
[INFO]: Final Vertical PDN Pitch: 29.286666666666665
[INFO]: Final Horizontal PDN Pitch: 29.013333333333335
[INFO]: Changing layout from 0 to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def
[INFO]: Running IO Placement...
[INFO]: current step index: 4
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/3-verilog2def_openroad.def to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def
[INFO]: Running Tap/Decap Insertion...
[INFO]: current step index: 5
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/tmp/floorplan/4-ioPlacer.def to /openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def
[INFO]: Power planning the following nets
[INFO]: Power: VPWR
[INFO]: Ground: VGND
[INFO]: Generating PDN...
[INFO]: current step index: 6
[INFO]: current step index: 7
[INFO]: PDN generation was successful.
[INFO]: Changing layout from /openlane/designs/spm/runs/openlane_test/results/floorplan/spm.floorplan.def to /openlane/designs/spm/runs/openlane_test/tmp/floorplan/6-pdn.def
[INFO]: Running Placement...
[... 111 lines cut ...]
[INFO]: Design Name: spm
Run Directory: /openlane/designs/spm/runs/openlane_test
----------------------------------------

Magic DRC Summary:
Source: /openlane/designs/spm/runs/openlane_test/reports/magic/36-magic.drc
Total Magic DRC violations is 0
----------------------------------------

LVS Summary:
Source: /openlane/designs/spm/runs/openlane_test/results/lvs/spm.lvs_parsed.lef.log
LVS reports no net, device, pin, or property mismatches.
Total errors = 0
----------------------------------------

Antenna Summary:
Source: /openlane/designs/spm/runs/openlane_test/reports/routing/38-antenna.rpt
Number of pins violated: 0
Number of nets violated: 0
[INFO]: check full report here: /openlane/designs/spm/runs/openlane_test/reports/final_summary_report.csv
[INFO]: There are no max slew violations in the design at the typical corner.
[INFO]: There are no hold violations in the design at the typical corner.
[INFO]: There are no setup violations in the design at the typical corner.
[SUCCESS]: Flow Completed Without Fatal Errors.
Basic test passed

closed 2021-11-07 by donn