The-OpenROAD-Project/OpenLane #694: Building docker image ignores build-arg

vvbandeira, 2021-11-07:
https://github.com/The-OpenROAD-Project/OpenLane/blob/18a17f40742879c88cd6393e540f148a8738102e/docker_build/docker/openroad_app/Dockerfile#L47-L48

I think these arguments need to be globally defined. See similar issue [here](https://github.com/moby/moby/issues/38379#issuecomment-448445652).

I have noticed that in our Jenkins pipeline we pass the correct commit hash:
```
[2021-11-07T08:26:30.103Z] docker build --rm --build-arg OPENROAD_APP_REPO=https://github.com/The-OpenROAD-Project/OpenROAD --build-arg OPENROAD_APP_COMMIT=ee6f9908d01de3d8cd716acb5b1e4ad234d26cdd -t openroad_app docker/openroad_app | tee logs/docker/openroad_app.build.txt
```

But during the build, the argument is ignored:
```
[2021-11-07T08:28:51.297Z] Step 9/20 : ARG OPENROAD_APP_COMMIT=4d4d7205fd0292dbf3fae55fad9109b3f0bd5786
```

Full log [here](https://jenkins.openroad.tools/blue/rest/organizations/jenkins/pipelines/OpenLane-Public-Test/runs/104/nodes/24/steps/26/log/?start=0).

(@maliberty FYI)

donn, 2021-11-07:
Can you verify that? The Docker log will always show the default value for the ARG when it's time to print the step. Please check the commit of the actual OpenROAD binary produced.

vvbandeira, 2021-11-07:
Yeah, sorry. I had a similar issue with build-args and just assumed here.
Looks like PR #630 broke our Jenkins script when `CURRENT_TAG -> OPENLANE_TAG`, but since make does not warn or error out when passing garbage the CI kept green.

closed 2021-11-07 by vvbandeira