The-OpenROAD-Project/OpenLane #695: Magic LEF/DEF Based DRC check fails where as GDS DRC check passes

dineshannayya, 2021-11-09:
### Description

Openlane support two way of MAGIC DRC check 1) LEF/DEF and 2) GDS using the environmental variable MAGIC_DRC_USE_GDS
I see GDS base DRC check passes, while as LEF/DEF based DRC fails with Error

### Environment
```
[issue.survey.rpt.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7501655/issue.survey.rpt.gz)

This issue can be reproduced with
1. Clone the https://github.com/efabless/caravel_user_project.git
2. Set set ::env(MAGIC_DRC_USE_GDS) "0"  in openlane/user_project_wrapper/config.tcl
3. cd openlane
4. make user_proj_example
5. make user_project_wrapper
Failure: Magic DRC fails with error

```

### Reproduction Material
1. make sure that $PDK_ROOT is configure to PDK path where Sky130A installed
2. Use Run command to reproduce the issue: run_cmd
[exp2.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7501614/exp2.tar.gz)

3. For Checking GDS mode of DRC checking, change File:  input/drc.tcl
set ::env(MAGIC_DRC_USE_GDS) "0"
           to
set ::env(MAGIC_DRC_USE_GDS) "1"
and run the run_cmd

### Expected behavior
As GDS based DRC is passing expectation is LVS/DEF also should pass.

### Logs

```
Log and Report are part of the example tar ball under output directory

```

RTimothyEdwards, 2021-11-10:
I would like to add that the specific problem here is that the OpenROAD router appears not to honor the wide-spacing rules between the routes and the obstruction layers defined in the macros.  There happen not to be any actual DRC errors because the obstruction layers extend out further than any of the internal routes, but that does not invalidate the issue, since the router should be maintaining a DRC distance per the information it sees in the macro LEF.

donn, 2022-06-23:
So there are two issues here: one has been filed to Open_PDKs. The other is an issue with OpenROAD generating a bad DEF, so only you can test that.

donn, 2023-04-10:
Issue was: https://github.com/The-OpenROAD-Project/OpenROAD/issues/2892

But fix is blocked on: https://github.com/The-OpenROAD-Project/OpenROAD/issues/3131

kareefardi, 2024-07-14:
Closing this as blocked PR is merged. If similar issues surfaces please open a new issue.

closed 2024-07-14 by kareefardi