The-OpenROAD-Project/OpenLane #696: run_time error during routing

vasanthramuk, 2021-11-09:
[temp_digital_or_droute_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7504510/temp_digital_or_droute_packaged.tar.gz)

### Description
I am using sky130 as PDK with OpenLane. During the flow, I encountered an error 

### Environment
```
Please run the following command in the extracted file: "./run  

python3 ./env.py issue-survey

And copy and paste the ENTIRE output between the triple-backticks.

If there's no env.py, you are using an out of date version of OpenLane and should probably update. 
```

### Reproduction Material
Please do ./run in the extracted directory

### Expected behavior
The flow should be successfull. But I am getting, 
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/or_droute.tcl |& tee >&@stdout /project/openlane/temp_digital/runs/temp_digital/logs/routing/24-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: SIGABRT

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /project/openlane/temp_digital/runs/temp_digital/error.log

### Logs
```
```

vijayank88, 2021-11-09:
I ran user source verilog code directly in openlane flow got stopped with below error:
```
[INFO DRT-0165] Start pin access.
[ERROR DRT-0073] No ap for _526_/SET_B.
terminate called after throwing an instance of 'std::runtime_error'
  what():  DRT-0073
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_droute.tcl |& tee >&@stdout /openLANE_flow/designs/temp_digital/runs/2ndrun/logs/routing/22-tritonRoute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: SIGABRT

```

vijayank88, 2021-11-10:
@Stephanommg  if you need source verilog code let me know. I will share with you. Thanks

dralabeing, 2021-11-10:
@vijayank88 You were able to run this successfully in ORFS right?

donn, 2021-11-10:
Tracking this in #618. This isn't an issue with OpenLane or OpenROAD per se, it's with the sky130 PDK.

closed 2021-11-10 by donn

Stephanommg, 2021-11-10:
I actually discovered the bug and it is because the router understands that all ports of a pin should have some access point, but this is not correct (AFAIK), we need at least one port with some access point.

maliberty, 2021-11-10:
@Stephanommg  This is a standard cell with multiple PORTS in LEF?

maliberty, 2021-11-10:
@donn how do you know this is the same as #618 ?

vijayank88, 2021-11-11:
> @vijayank88 You were able to run this successfully in ORFS right?

yes. I am able to complete the flow with ORFS

Stephanommg, 2021-11-11:
> @Stephanommg This is a standard cell with multiple PORTS in LEF?

yes

maliberty, 2021-11-11:
@Stephanommg what cell / pin is this?

Stephanommg, 2021-11-11:
inst: _523_
term: SET_B

maliberty, 2021-11-11:
File https://github.com/google/skywater-pdk/issues/343