The-OpenROAD-Project/OpenLane #697: rewrite or_replace.tcl to use global_placement command not low-level swig

maliberty, 2021-11-09:
### Description
In or_replace.tcl you should be using the global_placement command not mucking around with the swig layer.  The current code for PL_ROUTABILITY_DRIVEN is wrong and routability can't be turned off.  Please replace all this with a proper command invocation.

### Environment
latest

### Reproduction Material
any design

### Expected behavior
match OR

### Logs
None

antonblanchard, 2021-11-10:
I submitted https://github.com/The-OpenROAD-Project/OpenLane/pull/700 before I noticed this, which is obviously the better path to take.

FYI that isn't the only issue, @jjcherry56 points out the timing driven mode is missing wire RC values: https://github.com/The-OpenROAD-Project/OpenROAD/issues/1271#issuecomment-964694950

antonblanchard, 2021-11-11:
It looks like [or_groute.tcl](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/openroad/or_groute.tcl) should be updated as well.

jjcherry56, 2021-11-11:
I don't a problem in or_groute.tcl because it is using global routing parasitics.

PR #702 (2021-11-12): OpenROAD Script Updates
* Rewrite or_replace.tcl

That script used a lot of internal swig commands with an unstable API/unpredictable behavior. Resolves #697.

* Expose `max_displacement` flag for detailed placement

This was added at some point as a replacement to the diamond search height thing and is necessary for macros.

cc @mguthaus 

* Add OpenROAD GUI/Tcl Readline

Resolves #642. Note that this addition will be intentionally undocumented and is intended more for people who absolutely want it.

MISC:

* CI fast fail off/always push image

The CI will report failures still, but will stil [... cut]
commit 8950f1aa (2021-11-12): OpenROAD Script Updates (#702)

* Rewrite or_replace.tcl

That script used a lot of internal swig commands with an unstable API/unpredictable behavior. Resolves #697.

* Expose `max_displacement` flag for detailed placement

This was added at some point as a replacement to the diamond search height thing and is necessary for macros.

* Add OpenROAD GUI/Tcl Readline

Resolves #642. Note that this addition will be intentionally undocumented and is intended more for people who absolutely want it.

MISC:

* CI fast fail off/always push image

The CI will report failures still, but will still publi [... cut]