The-OpenROAD-Project/OpenLane #710: Flow crushed at STA Multicorner - segmentation fault in the newest docker image tag: 2021.11.13_01.43.55

KHermanUBB, 2021-11-13:
### Description
Happy to see that there is a multi-corner analysis however it fails on the SS corner 

### Environment
```
Python: v3.6.9
Kernel: Linux v4.15.0-162-generic
Distribution: ubuntu 18.04
Container Engine: docker v20.10.7
OpenLane Git Version: 2021.11.13_01.43.55

Alert: Click is not installed.
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

commit cbb562bd43c5c410b1b498604803c3dd88a44856
Author: Donn <me@donn.website>
Date:   Sat Nov 13 13:20:00 2021 +0200

    Assorted Usability Improvements (#709)
    
    * verilog2def -> init_floorplan
    
    Some files still retained the name of a floorplanning tool we're no longer using. It has been removed.
    
    * netlist locations
    
    Netlists generated as part of later steps were placed under `synthesis/`. They have been moved to their proper locations.
    
    * Build tweaks
    
    `make openlane` now actually builds the container. `make pull-openlane` has been added with the original functionality.
    
    The git commit has been placed in the container at `/git_version`.
    
    A custom `.bashrc` has been added.
    
    Non-standard `echo -ne` invocations replaced with `printf`.
    
    * Issues
    
    verify_versions now works without a `PDK_ROOT` set (`$openlane_root/pdks`)
    
    clarified issue template

commit 9958a53dd085020bbc468ea1ae663da57d19357f
Author: Anton Blanchard <anton@linux.ibm.com>
Date:   Sat Nov 13 15:52:06 2021 +1100

    Update documentation to show RUN_SIMPLE_CTS defaults to off (#708)

commit 9d2c4de07e93d9a8f520c3e8d6be896d6993cee6
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Fri Nov 12 15:14:54 2021 -0600

    Bring back `FP_IO_MIN_DISTANCE` (#705)
    
    * Bring back FP_IO_MIN_DISTANCE
    
    * Set io min distance default to 3

```

### Reproduction Material
https://github.com/KHermanUBB/cup-eco.git

### Expected behavior
Clean pass 

### Logs
```
======================= Slowest Corner ===================================
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/sta_multi_corner.tcl |& tee >&@stdout /project/openlane/user_proj_example/runs/user_proj_example/logs/routing/25-spef_extraction_multi_corner_sta"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation
```

donn, 2021-11-13:
Thank you for the report- we'll investigate.

donn, 2021-11-24:
I cloned your repo. Works fine on master.

closed 2021-11-24 by donn