The-OpenROAD-Project/OpenLane #718: Support for overriding the module parameter during synthesis

dineshannayya, 2021-11-17:
### Prompt
When Generic IP which has multiple parameter over-ride option, it would be better to provide module parameter over-ride during synthesis.

Yosys support <chparam> command to overide the module parameter.  
#### HERE is Yosys Example command ######################
# Memory bits <= 18K; Data width <= 36; Address width <= 14:
read_verilog ../common/blockram.v
chparam -set ADDRESS_WIDTH 10 -set DATA_WIDTH 1 blockram

This module parameter option is not supported in Openlane flow.
### Proposal
Add additional environmental variable like : SYNTH_PARAMS

set ::env(SYNTH_PARAMS) "BIST_ADDR_WD 8,\
	                 BIST_DATA_WD 32,\
		         BIST_ADDR_START 9'h000,\
			 BIST_ADDR_END 9'h1FB,\
			 BIST_REPAIR_ADDR_START 9'h1FC,\
			 BIST_RAD_WD_I 9,\
			 BIST_RAD_WD_O 9\
			 "
Add the following logic in synth.tcl & synth_top.tcl
## Module level parameter overide 
if { [info exists ::env(SYNTH_PARAMS) ] } {
    log "Reading $::env(SYNTH_PARAMS) as a parameter"
    set records [split $::env(SYNTH_PARAMS) ","]
    foreach rec $records {
       chparam -set [lindex $rec 0] [lindex $rec 1] $vtop
    }
} else {
	log "No parameter define found"
}

[synth.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/7556024/synth.txt)
[synth_top.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/7556025/synth_top.txt)

Attached the modified scripts/synth.tcl and scripts/synth_top.tcl 
Note: File name extension is changes from .tcl to .txt due to github attach error/

### Alternatives
Currently we have to manually hardcode the paremeter inside the generic module.

donn, 2021-11-23:
Agreed. It would speed things up if you create a PR, but this is definitely a much-needed enhancement.

dineshannayya, 2022-08-01:
I see this feature is supported in latest openlane tool set

closed 2022-08-01 by dineshannayya

scorbetta, 2024-01-24:
Hello all, I'm having issues using the `SYNTH_PARAMETERS` option.

According to [this](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/reference/configuration.md) the `SYNTH_PARAMETERS` option is of the form:

```
// From: config.tcl
set ::env(SYNTH_PARAMETERS) "NUM_INPUTS=4 WIDTH=8 FRAC_BITS=5"
````

However, when I use the above, the synthesis always complains about a lot of ports having no drivers, then trims them and I get an empty design which eventually makes the synthesis exit with an error. Please note that I have already checked those ports and all wires are properly connected. So those warnings are wrong. However, if I remove the above `SYNTH_PARAMETERS` option and let the default `parameter` values in my Verilog code (with values exactly equal to the above), then everything goes as expected.

What am I missing? Is there a problem with my syntax above?

Many thanks
S