The-OpenROAD-Project/OpenLane #733: Artifact reorganisation creates bad path during synthesis

antonblanchard, 2021-11-24:
After updating Openlane I hit the following error during synthesis:

```
ERROR: Can't create file /project/openlane/user_project_wrapper/runs/user_project_wrapper/reports/synthesis/1-synthesis.log/2-synthesis_pre.stat.
```

synth_report_prefix and synthesis_reports are indeed bad:

```
set ::env(synth_report_prefix) {/project/openlane/user_project_wrapper/runs/user_project_wrapper/reports/synthesis/1-synthesis.log/2-synthesis}
set ::env(synthesis_reports) {/project/openlane/user_project_wrapper/runs/user_project_wrapper/reports/synthesis/1-synthesis.log/2-synthesis.log}
```

It does look like an issue if we call run_yosys twice:

```
        set ::env(synth_report_prefix) [index_file $::env(synthesis_reports)/synthesis]
        set ::env(synthesis_reports) [index_file $::env(synthesis_reports)/synthesis.log]
```

donn, 2021-11-24:
So er, wait, I'm confused. Does synthesizing twice cause this issue?

antonblanchard, 2021-11-24:
I presume it's because my design has this:

```
set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"
```

Which makes us call `run_yosys` a second time:

```
        if { [info exists ::env(SYNTH_USE_PG_PINS_DEFINES)] } {
                puts_info "Creating a netlist with power/ground pins."
                if { ! [info exists ::env(SYNTH_DEFINES)] } {
                        set ::env(SYNTH_DEFINES) [list]
                }
                lappend ::env(SYNTH_DEFINES) {*}$::env(SYNTH_USE_PG_PINS_DEFINES)
                run_yosys -output $::env(synthesis_tmpfiles).pg_pins.v -no_set_netlist
        }
```

donn, 2021-11-25:
Gotcha. I'll take a look.

donn, 2021-11-25:
Okay, sorry- oversight on my part. `synthesis_reports` should never be re-set. PR created, and it should close this issue once merged.

I'm terribly sorry for the inconvenience.

commit f20cdb39 (2021-11-25): Fix #733 (#739)
closed 2021-11-25 by donn