The-OpenROAD-Project/OpenLane #735: DRC violations

rahulrbharadwaj, 2021-11-24:
### Description
I have an analog module which I have layed out in hand using the latest version of magic with no DRC violations. I have called this analog module as a macro in openlane and the ran the flow. At the end of the flow there are DRC violations. 

### Environment
```
PDK Version Verification Status: OK

```

### Reproduction Material
[https://drive.google.com/drive/folders/1qruuyWZgdvJDj1c_m4O1XuFW0_pasv9c?usp=sharing](url)

### Expected behavior
There must be no DRC violations.

### Logs
```
The log file is included in the tar compressed file.
```

vijayank88, 2021-11-25:
@rahulrbharadwaj 
follow instruction properly and add your tar file to reproduce the issue.
Given Reproduction Material link is empty.

rahulrbharadwaj, 2021-11-25:
@vijayank88 can you view the files now?

donn, 2021-12-01:
Your environment survey is also pretty empty, only including the PDK part and nothing else. We need the whole thing.

closed 2021-12-01 by rahulrbharadwaj