The-OpenROAD-Project/OpenLane #737: make pull-openlane failed

vijayank88, 2021-11-25:
### Description
Not able to update latest openlane commit.

### Environment
```
python3 ./env.py issue-survey
Python: v3.8.10
Kernel: Linux v5.11.0-40-generic
Distribution: ubuntu 20.04
Container Engine: docker v20.10.10
OpenLane Git Version: 2021.11.25_01.26.14
---
PDK Version Verification Status: MISMATCH
The version of open_pdks installed does not match the one required by the OpenLane flow scripts (installed: 14db32aa8ba330e88632ff3ad2ff52f4f4dae1ad, expected: c5730b574461889c82858b08d12ba42423d9c2cb)
You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

commit 287cb3a712f1421097794e480177835383766187
Author: Kareem Farid <kareefardi@users.noreply.github.com>
Date:   Wed Nov 24 21:46:17 2021 +0200

    Verify existence of EXTRA_GDS_FILES (#731)

commit 520f2f136a1cd3e7b99485740b724dadc974eecf
Author: Donn <me@donn.website>
Date:   Wed Nov 24 14:59:08 2021 +0200

    Update OpenROAD, Magic and CVC (#734)
    
    CVC now also points at master instead of @donn's fork

commit 65da365962a942b595d59b64a5dbdbc550cd91cc
Author: Anton Blanchard <anton@linux.ibm.com>
Date:   Wed Nov 24 23:50:31 2021 +1100

    Fix typos in PL_RESIZER_REPAIR_TIE_FANOUT option (#732)
    
    [skip ci]

```

### Reproduction Material
```
    cd openlane/
    git checkout master
    git pull
    export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
    make pull-openlane
```

### Expected behavior
complete make pull-openlane complete successfully. 

### Logs
```
  File "./dependencies/get_tag.py", line 26
    process_data: subprocess.CompletedProcess = subprocess.run(["git", "describe", "--tags", "--abbrev=0"], stdout=subprocess.PIPE, stderr=subprocess.PIPE)
                ^
SyntaxError: invalid syntax
Pulling most recent OpenLane image relative to your commit...
docker pull 
"docker pull" requires exactly 1 argument.
See 'docker pull --help'.

Usage:  docker pull [OPTIONS] NAME[:TAG|@DIGEST]

Pull an image or a repository from a registry
make: *** [Makefile:184: pull-openlane] Error 1

```

vijayank88, 2021-11-26:
@donn even after sed its not worked. By setting IMAGE_NAME works.
```
export OPENLANE_IMAGE_NAME=efabless/openlane:2021.11.26_01.21.38
```

donn, 2021-11-26:
Yeah, sorry, I have no idea how this is happening. The only thing that makes sense is an out of date python version.

maliberty, 2021-11-26:
@donn I'm not familiar with this python syntax "process_data: subprocess.CompletedProcess = subprocess.run" what does it mean?

donn, 2021-11-26:
The `: subprocess.CompletedProcess` is a typing hint. [PEP-0484](https://www.python.org/dev/peps/pep-0484/), implemented in Python 3.5+, allows for type hints to assist IDE or other utilities with autocomplete and/or enforcing some basic type safety.

The idea is, a typical Python interpreter would read and ignore this part: so functionally this would be `process_data = subprocess.run(...)`. However, if you're running a version of Python where the AST simply doesn't recognize type hints, they report a syntax error.

maliberty, 2021-11-26:
@donn Perhaps you should check sys.version_info to ensure you have the minimal required version.

donn, 2021-11-26:
I do. As part of the environment survey that I ask people to run before submitting any issue, which is reporting Python 3.8 here.

vijayank88, 2021-11-27:
@donn 
In Makefile default python3 used. If I change it to python3.8 everything working fine.
I've 2.7,3.5,3.8 version in Ubuntu. As openlane supports 3.5+ above, change to 3.8 works for me.

donn, 2021-11-27:
I can't do that. People's setups range from Python 3.6 (centos) to Python 3.9 (my setup). Changing it to python3.8 would break unless someone specifically had Python3.8 installed. Not acceptable.

vijayank88, 2021-11-29:
@donn  as problem comes from my python version, fixed myself and closing this issue.

closed 2021-11-29 by vijayank88