The-OpenROAD-Project/OpenLane #750: Failure at cts step

ratul619, 2021-12-08:
Hi , 

While running any design , see that cts is getting exited with below message : 

-----------------------------------------------------------------------

set_driving_cell -lib_cell sky130_fd_sc_hd__inv_2 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {write_data[2]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_2 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {write_data[1]}]
set_driving_cell -lib_cell sky130_fd_sc_hd__inv_2 -pin {Y} -input_transition_rise 0.0000 -input_transition_fall 0.0000 [get_ports {write_data[0]}]
set_timing_derate -early 0.9500
set_timing_derate -late 1.0500
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 8.0000 [current_design]
[INFO]: Setting RC values...
[INFO]: Configuring cts characterization...
[INFO]: Performing clock tree synthesis...
[INFO]: Looking for the following net(s): clk
[INFO]: Running Clock Tree Synthesis...
[ERROR]: during executing: "./openroad -exit /home/ratul619/Desktop/OpenLane/scripts/openroad/cts.tcl |& tee >&@stdout /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/logs/cts/11-cts.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check ./openroad  log file
[ERROR]: Dumping to /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: aes
Run Directory: /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33
Source not found.
----------------------------------------

LVS Summary:
Source: /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/logs/finishing/aes.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
[ERROR]: Flow failed.

    while executing
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/cts.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(cts_logs)/cts.log]"
    (procedure "run_cts" line 27)
    invoked from within
"run_cts"
    (procedure "run_cts_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 54)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 360)

-----------------------------------------------------------------------
The error.log has only below : 

ratul619@ratul619-VirtualBox:~/Desktop/OpenLane$ cat /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/error.log 
during executing: "./openroad -exit /home/ratul619/Desktop/OpenLane/scripts/openroad/cts.tcl |& tee >&@stdout /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/logs/cts/11-cts.log"
Last 10 lines:
child killed: segmentation violation
----------------------------------------------------------------
The cts.log : 

ratul619@ratul619-VirtualBox:~/Desktop/OpenLane$ tail  /home/ratul619/Desktop/OpenLane/designs/aes/runs/RUN_2021.12.08_10.34.33/logs/cts/11-cts.log 
set_timing_derate -late 1.0500
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 8.0000 [current_design]
[INFO]: Setting RC values...
[INFO]: Configuring cts characterization...
[INFO]: Performing clock tree synthesis...
[INFO]: Looking for the following net(s): clk
[INFO]: Running Clock Tree Synthesis...
-------------------------------------------------

Any clue what is going wrong there ? 

Thanks!

vijayank88, 2021-12-08:
@ratul619 
run
```
 python3 ./env.py issue-survey
```
provide the details

ratul619, 2021-12-19:
Using docker - issue is no longer there/

closed 2021-12-19 by ratul619