The-OpenROAD-Project/OpenLane #751: Fail to create base image

vvbandeira, 2021-12-08:
### Description
`make -C docker base_image` fails on the master branch.

### Environment
```
Python: v3.6.8
Kernel: Linux v3.10.0-1160.42.2.el7.x86_64
Distribution: centos 7
Container Engine: docker v20.10.9
OpenLane Git Version: 2021.12.08_01.36.24

Alert: Click is not installed.
---
PDK Version Verification Status: FAILED
/home/vvbandeira/openlane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/vvbandeira/openlane/dependencies/verify_versions.py", line 120, in verify_versions
    raise Exception(f"{sky130_dir} not found.")
Exception: /home/vvbandeira/openlane/pdks/sky130A not found.

Failed to compare PDKs.
---
Git Log (Last 3 Commits)

commit 537e4329875f85709f5316e48c7ad60ef98f890d
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Mon Dec 6 23:23:25 2021 -0600

    Add variable for enabling -post_cts_disable option (#749)

commit 445acc69b3a1180533f7133fb1db08ab4fdedd9a
Author: Donn <me@donn.website>
Date:   Thu Dec 2 10:55:01 2021 +0200

    Report Generation Bugfixes (#742)

    * Report Generation Bugfixes

    + Added `-override_env` option to flow.tcl to override certain environment variables on a per-run basis
    + Added option to extract inverter cells into the metrics behind an env var (so as not to break compatibility)
    + Klayout requirements added to runtime
    ~ Regression test outputs renamed, design report now in yaml
    ~ CI now runs on ubuntu-20.04 specifically, installs pyyaml
    ~ Fixed `make test`
    ~ Rewrote compare_regression_reports in click
    ~ Fixed compare_regression_design (typo made it not run, aliasing failures!!)
    ~ Changes to variable names: FINAL_TIMING_REPORT_TAG is now LAST_TIMING_REPORT_TAG, ran every STA
    ~ Fixed yosys gate type extraction

    * Fix Odd Antenna Issue By Reverting OpenROAD

commit e3fc1f5d0e0187032b81e05de4811ab611ed9bd6
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Tue Nov 30 08:50:53 2021 -0600

    Save SPEF/SDC/SDF files after the flow finishes (#741)
```

### Reproduction Material
Unmodified master branch.

### Expected behavior
Generate base image without error.

### Logs
```
Step 5/16 : RUN yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_precompile_dependencies.txt | tr '\n' ' ') && 	yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_compile_dependencies.txt | tr '\n' ' ') && 	yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_dependencies.txt | tr '\n' ' ') && 	yum clean all && 	rm -rf /var/cache/yum
 ---> Running in 0d10ee41b4c0
[...]
Package 1:qt-4.8.7-9.el7_9.x86_64 already installed and latest version
Package qt5-qtbase-5.9.7-5.el7_9.x86_64 already installed and latest version
Package qt-settings-19-23.12.el7.centos.noarch already installed and latest version
Package 1:qt-x11-4.8.7-9.el7_9.x86_64 already installed and latest version
Package ruby-2.0.0.648-36.el7.x86_64 already installed and latest version
No package rubygem available.
Error: Not tolerating missing names on install, stopping.
The command '/bin/sh -c yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_precompile_dependencies.txt | tr '\n' ' ') &&    yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_compile_dependencies.txt | tr '\n' ' ') &&       yum install --setopt=skip_missing_names_on_install=False -y $(cat /yum_dependencies.txt | tr '\n' ' ') &&       yum clean all &&        rm -rf /var/cache/yum' returned a non-zero code: 1
make: Leaving directory `/home/vvbandeira/openlane/docker'
```
Complete log file: [log.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/7676755/log.txt)

@maliberty FYI

maliberty, 2021-12-08:
@donn this breaks our OL CI test so please give it priority.

donn, 2021-12-08:
here u go

commit 6ad621ea (2021-12-08): Fix #751 (#752)
closed 2021-12-08 by donn