The-OpenROAD-Project/OpenLane #753: Changes in odb API require script updates in OL

maliberty, 2021-12-08:
### Description
In https://github.com/The-OpenROAD-Project/OpenROAD/pull/1392 there were some API changes in odb.  This means that code that code will need to be update for dbITerm connect/disconnect:

odb.dbITerm_disconnect(iterm)  -> iterm disconnect
odb.dbITerm_connect(iterm, net) -> iterm connect net

This occurs in contextualize.py, dont_buffer.py, label_macro_pins.py, place_diodes.py, power_route.py, write_powered_def.py, openroad/diodes.tcl based on grep'ing.

### Environment
head of master

### Reproduction Material
Fails in make test 

### Expected behavior
doesn't fail

### Logs
09:54:03  Traceback (most recent call last):
09:54:03    File "/openlane/scripts/write_powered_def.py", line 182, in <module>
09:54:03      odb.dbITerm_connect(VDD_ITERM, DEFAULT_VDD)
09:54:03  AttributeError: module 'odb' has no attribute 'dbITerm_connect'
09:54:03  child process exited abnormally

[This is failing the OR CI test for OL]

maliberty, 2021-12-08:
@vvbandeira FYI

donn, 2021-12-09:
I'm supposed to be off this week, y'all. Can you try your hand at fixing these?

maliberty, 2021-12-09:
I didn't realize you are off.  It can wait or you can reassign it.  I'm not sure how changing this in master works as it depends on an OR update as well.

donn, 2021-12-09:
The idea is you create a branch where you update the OpenROAD app and the variables together, then PR that.

donn, 2021-12-14:
Fixed in #756.

closed 2021-12-14 by donn