The-OpenROAD-Project/OpenLane #754: Segfault in one of OpenLane's steps (or_replace.tcl)

progirep, 2021-12-09:
### Description
I'm getting a segfault in one of OpenLane's steps, but can't see what a possible reason for this could be (sorry, I don't "speak" TCL nor do I use docker for my own work - I'm only doing a few assembler styles, C, C++, Python, Java, some HDL, and CS Theory).

The project on which the problem occurs is a test project for integrating OpenRAM blocks into a design - it is a shortened version of the project at https://github.com/VLSIDA/openram_testchip.git using only two SRAM blocks.

### Environment
The problem occurs both with the mpw-3a version of the workflow as well as with the "latest" (b911a5bbbcaa) version of the OpenLane docker images (selected by changing the respective lines in the files "Makefile" and "openlane/Makefile"). In the case of the "latest" version latter case, the gds.gz files in the gds/ directory of the project need to be gunzipped for the flow to reach the segfault. Also, when using the "latest" version, there is a warning for me that the PDK version is then not the right one - not sure if this is relevant because the segfault is the same.

```
Python: v3.8.10
Kernel: Linux v5.4.0-91-generic
Distribution: ubuntu 20.04
Container Engine: docker v20.10.7
OpenLane Git Version: 2021.10.25_20.35.00
---
PDK Version Verification Status: OK

---
Git Log (Last 5 Commits)

commit e8f4a88f668b366f126bba40861153bf478a33c1
Author: Donn <me@donn.website>
Date:   Mon Oct 25 20:32:41 2021 +0200

    Last Minute Fixes for MPW3 (#675)

```

### Reproduction Material

Github doesn't let me upload the project as a tar.gz file or a .zip file with the message "is not included in the list". Maybe it is too large. So I've committed the project as a project to github at: https://github.com/progirep/openram_testchip/tree/8ae25c1d4a85146cf2a7d277da1f87e1234a4b85

The following zip file is just a dummy to be able to submit the issue anyway: 

The command to run is simply:
```
make user_project_wrapper
```
from the project directory.

### Expected behavior
The workflow should not segfault. The project I'm trying to run it on *may* be wrong in some aspects, but right now it's not possible for me to tell because of the segfault.

### Logs
```
...[stuff that seems to work]....
[INFO GRT-0104] Minimal overflow 1 occurring at round 0.
[INFO GRT-0111] Final number of vias: 3700
[INFO GRT-0112] Final usage 3D: 89696
*** Error in `openroad': corrupted double-linked list: 0x0000000014d8d870 ***
======= Backtrace: =========
/lib64/libc.so.
[test.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/7683839/test.zip)
6(+0x7f3e4)[0x7f47d70043e4]
/lib64/libc.so.6(+0x82ba8)[0x7f47d7007ba8]
/lib64/libc.so.6(__libc_malloc+0x4c)[0x7f47d700a6fc]
/lib64/libstdc++.so.6(_Znwm+0x1d)[0x7f47d78ca18d]
openroad[0xc3ac4b]
openroad[0x10293e2]
openroad[0x102afd9]
openroad[0xc349ab]
openroad[0xc38e6a]
openroad[0x51d454]
openroad[0x4fad98]
openroad[0x525ad1]
/lib64/libtcl8.5.so(+0x34eb2)[0x7f47d892feb2]
/lib64/libtcl8.5.so(+0x7936c)[0x7f47d897436c]
/lib64/libtcl8.5.so(+0x81647)[0x7f47d897c647]
/lib64/libtcl8.5.so(TclEvalObjEx+0x87)[0x7f47d89316b7]
/lib64/libtcl8.5.so(+0xbc27f)[0x7f47d89b727f]
[... 154 lines cut ...]
"flow_fail"
    (procedure "try_catch" line 25)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -exit $::env(SCRIPTS_DIR)/openroad/or_replace.tcl |& tee $::env(TERMINAL_OUTPUT) [index_file $::env(replaceio_log_file_..."
    (procedure "global_placement_or" line 14)
    invoked from within
"global_placement_or"
    (procedure "run_placement" line 17)
    invoked from within
"run_placement"
    (procedure "run_placement_step" line 11)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 43)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "/openlane/flow.tcl" line 356)
make[1]: *** [Makefile:43: user_project_wrapper] Error 1
make[1]: Leaving directory '/home/...(my user name and install path here).../openram_based_monitoring/openlane'
make: *** [Makefile:83: user_project_wrapper] Error 2
```

vijayank88, 2021-12-09:
@progirep mpw-3b is right version for caravel project. Try with mpw-3b tag

progirep, 2021-12-09:
> @progirep mpw-3b is right version for caravel project. Try with mpw-3b tag

Thanks, I will try that and report back in ~2-8 hours.

donn, 2021-12-09:
> sorry, I don't "speak" TCL

vijayank88, 2021-12-09:
@donn  this issue reproduced with mpw-3b tag as well

donn, 2021-12-09:
@vijayank88 Really not much I can do other than or_issue the reproducible and try it with the latest version of OpenROAD. However, I'm off this week, so that'll have to wait until Monday.

progirep, 2021-12-09:
Ok, I've just checked with the "mpw-3b" version of the docker container. The segfault occurs as well, and the stack trace is the same up to the memory addresses (which are unlikely to be the same because of address space randomization).

The problem does not occur for the design from which my test design is forked. I'll have a look if there is something to be seen from the differences between the OpenLane warnings - in case the segfault only occurs for designs that have no chance to work anyway, there may be some useful information hidden to track down the problem.

donn, 2021-12-09:
@progirep This failure occurs inside the OpenROAD app specifically, so there's not much we can do on our side but package the issue and pass it on to the OpenROAD team. Likely there's a corner case exposed by your design that's just not handled properly.

vijayank88, 2021-12-10:
@progirep 
follow the instruction in https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md and provide tar file to resolve it

vijayank88, 2021-12-10:
@progirep segfault error comes openlane/Makefile.
Refer https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1636840104299800?thread_ts=1636753036.268100&cid=C016H8WJMBR.
Solution is :+1: 
change /openLANE_flow  to /openlane in openlane/Makefile

progirep, 2021-12-11:
Ok, so I've attached the tar file for reproducing the issue built according to the using_or_issue.md document: 
[user_project_wrapper_or_replace_packaged.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7697883/user_project_wrapper_or_replace_packaged.tar.gz)

The proposal to change the Makefile didn't work for me. It had no effect on the error. If I get it correctly, the change is for "injecting" the local OpenLane checkout from OPENLANE_ROOT into the docker container. I was using the "mpw-3a" version for that previously (which seems to be the one coming from this pull request: https://github.com/The-OpenROAD-Project/OpenLane/pull/675). Updating to a newer version of leads to the whole build process failing with "[ERROR STA-0402] repair_design -slew_margin is not a known keyword or flag.".

I guess that the slack link was posted for internal use because signing up requires e-mail addresses from few domains (google but not gmail, efabless, ..)

maliberty, 2021-12-12:
Your packaged testcase suggests you are using an old version of openlane.  What commit are you on?

donn, 2021-12-13:
I can't reproduce the issue with a later version of OpenROAD. Meaning that this issue is waiting on #756.

donn, 2021-12-13:
No problem! Just happy we could help.

closed 2021-12-13 by donn