The-OpenROAD-Project/OpenLane #755: `run_designs.py` should exit with a nonzero code if a design fails

donn, 2021-12-09:
### Prompt
`run_designs.py`, `make test_design_list` and its derivatives, `make fastest_test_set` and `make extended_test_set`, all exit with a zero exit code even if any of the designs fail. This aliases failures on the OpenROAD team's Jenkins CI, as they don't use `.github/scripts/run_tests.py`.

### Proposal
A flag should be added to `run_designs.py` for a nonzero exit, and this flag should be used by the Makefile.

vvbandeira, 2021-12-09:
@maliberty FYI, this will fix the false positives you flagged.

mithro, 2021-12-10:
Is there any reason you would want to keep the exisiting zero exit code on failure behaviour?

donn, 2021-12-12:
run_designs is definitely used in other places in the codebase that do not expect this behavior, and while I intend to alter these as well, I'm unsure if run_designs is also used by other codebases. A flag is easier than a full audit.

PR #760 (2021-12-14): Rewrite `run_designs.py`
+ run_designs.py now exits with exit code 2 if a design failed
~ run_designs.py rewritten in click, API altered slightly
~ renamed `addComputedStatistics` to `add_computed_statistics`
- removed `make regression_test`

Resolves #755