The-OpenROAD-Project/OpenLane #770: How can I pass verilog header files to OpenLane

microSharjeel, 2021-12-19:
### Prompt
A clear and concise description of what shortcoming you feel OpenLane has.
How can I pass verilog header files to OpenLane. OpenLane should recognize the .vh files as verilog header files.
Kindly, guide me how to add the verilog header files in OpenLane in interactive mode.
### Proposal
A clear and concise description of what you want to happen.
I have no idea
### Alternatives
A clear and concise description of any alternative solutions or features you've considered.

vijayank88, 2021-12-20:
@microSharjeel 
In the respective verilog file you just include 
```
`include "<file_name>.vh"
```
Yosys will handle automatically.

donn, 2021-12-20:
Correct. We currently don't have any special way of treating verilog headers... You just \`include them.

closed 2021-12-20 by donn

microSharjeel, 2021-12-20:
thanks for help it is resolved 
I am stating the solution for anyone who needs help!
To include the Verilog headers just 
include all verilog source files in VERILOG_FILES variable in config.tcl and 
include them in verilog source files and put the header files in same src folder as the .v files are. yosys will find them.
cheers!