The-OpenROAD-Project/OpenLane #772: OpenLane "child process exited abnormally"

jaquerinte, 2021-12-19:
### Description
When I launch my project, OpenLane crash with the error "child process exited abnormally".
Note that in my config.tlc. The CLOCK_PORT and CLOCK_NET is the same port, because, yosys simplifies the net.

### Environment
```
0 1
2 3
Python: v3.8.10
Kernel: Linux v5.11.0-43-generic
Distribution: ubuntu 20.04
Container Engine: docker v20.10.11
OpenLane Git Version: 2021.10.25_20.35.00
---
PDK Version Verification Status: OK

---
Git Log (Last 5 Commits)

commit e8f4a88f668b366f126bba40861153bf478a33c1
Author: Donn <me@donn.website>
Date:   Mon Oct 25 20:32:41 2021 +0200

    Last Minute Fixes for MPW3 (#675)

```

### Reproduction Material
* https://github.com/jaquerinte/space_controller/tree/4de4e18cf8117fdb59c9daad2f8f1f983a1ef04c
* command:
make user_proj_example

### Expected behavior
Openlane finish the hardening.

### Logs
```
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[ERROR GRT-0169] Invalid index for position (86, 81).
Error: or_resizer_routing_timing.tcl, 41 GRT-0169
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/or_resizer_routing_timing.tcl |& tee >&@stdout /project/openlane/user_proj_example/runs/user_proj_example/logs/routing/15-resizer_timing_optimization.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /project/openlane/user_proj_example/runs/user_proj_example/error.log
```

donn, 2021-12-20:
Can you please try on the current version of OpenLane?

jaquerinte, 2021-12-21:
Ok, now in the new version is fix. 
Thanks

closed 2021-12-21 by jaquerinte