The-OpenROAD-Project/OpenLane #773: Run stopping at global placement for blackbox based macro run :

ratul619, 2021-12-20:
Hi , 

I am just running blackbox based run - where one instance in the verilog is blackbox with .lib and .lef. 
The XYZ is just connected to the  inout pins.  So added below on confile file : 

-------------------------------------------------------------------
# Design
set ::env(DESIGN_NAME) "xyz_wrapper"

set ::env(VERILOG_FILES) [glob ./designs/xyz_wrapper/src/*.v]

set ::env(CLOCK_PERIOD) "10.000"
#set ::env(CLOCK_PORT) "clk"
set ::env(CELL_PAD) 4

#set ::env(FP_PDN_VPITCH) 1

set ::env(DIE_AREA) {0.0 0.0 500 500 }

set ::env(FP_SIZING) "absolute"

set ::env(SYNTH_READ_BLACKBOX_LIB) 1 

set ::env(LIB_SYNTH_COMPLETE_NO_PG) "./designs/rram_wrapper/XYZ.lib"
set ::env(EXTRA_LEFS) {./designs/rram_wrapper/XYZ.lef}

set ::env(STA_REPORT_POWER) {0}

#set ::env(TAP_DECAP_INSERTION) {0}

set ::env(QUIT_ON_TR_DRC) {0}

set ::env(ROUTING_OPT_ITERS) {10}
 

# set ::env(FP_PIN_ORDER_CFG) $::env(OPENLANE_ROOT)/designs/counter/pin_order.cfg

set filename ./designs/$::env(DESIGN_NAME)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
------------------------------------------------------------------------------------------

But flow is stopping : 

[INFO ODB-0131]     Created 3 components and 45 component-terminals.
[INFO ODB-0133]     Created 19 nets and 21 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/xyz/runs/RUN_2021.12.20_10.40.30/tmp/floorplan/4-io.def
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/replace.tcl |& tee >&@stdout /openlane/designs/xyz/runs/RUN_2021.12.20_10.40.30/logs/placement/5-global.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openlane/designs/xyz/runs/RUN_2021.12.20_10.40.30/error.log
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: xyz
Run Directory: /openlane/designs/xyz/runs/RUN_2021.12.20_10.40.30
Source not found.

--------------------- 
Verilog file : 

module xyz( in_X_0_0_top,
in_X_1_0_top,
in_X_2_0_top,
in_X_3_0_top,
in_X_4_0_top,
in_X_5_0_top,
in_X_6_0_top,
in_X_7_0_top,
in_X_0_1_top,
in_X_1_1_top,
in_X_2_1_top,
in_X_3_1_top,
in_X_4_1_top,
in_X_5_1_top,
in_X_6_1_top,
in_X_7_1_top,test,clk);

inout in_X_0_0_top;
inout in_X_1_0_top;
inout in_X_2_0_top;
inout in_X_3_0_top;
inout in_X_4_0_top;
inout in_X_5_0_top;
inout in_X_6_0_top;
inout in_X_7_0_top;
inout in_X_0_1_top;
inout in_X_1_1_top;
inout in_X_2_1_top;
inout in_X_3_1_top;
inout in_X_4_1_top;
inout in_X_5_1_top;
inout in_X_6_1_top;
inout in_X_7_1_top;
input clk;
output test;

XYZ U_sky130_fd_pr___xyz (
.in_X_0_0(in_X_0_0_top),
.in_X_1_0(in_X_1_0_top),
.in_X_2_0(in_X_2_0_top),
.in_X_3_0(in_X_3_0_top),
.in_X_4_0(in_X_4_0_top),
.in_X_5_0(in_X_5_0_top),
.in_X_6_0(in_X_6_0_top),
.in_X_7_0(in_X_7_0_top),
.in_X_0_1(in_X_0_1_top),
.in_X_1_1(in_X_1_1_top),
.in_X_2_1(in_X_2_1_top),
.in_X_3_1(in_X_3_1_top),
.in_X_4_1(in_X_4_1_top),
.in_X_5_1(in_X_5_1_top),
.in_X_6_1(in_X_6_1_top),
.in_X_7_1(in_X_7_1_top));

assign test = in_X_0_0_top &  in_X_1_0_top ; 

endmodule

-------------------------------------- 
LEF for XYZ bbox : 

VERSION 5.4 ;
NAMESCASESENSITIVE ON ;
BUSBITCHARS "[]" ;
DIVIDERCHAR "/" ;
UNITS
  DATABASE MICRONS 1000 ;
END UNITS
MACRO XYZ
    ORIGIN 0.0 0.0 ;
    CLASS BLOCK ;
    SIZE 7.4 BY 7.4 ; 
    SYMMETRY X Y R90 ;
    PIN  in_Y_0_0
         DIRECTION  INOUT ; 
         PORT 
             LAYER met2 ;
             RECT     0 0.65 0.5 1.15 ;
         END
    END  in_Y_0_0
    PIN  in_Y_1_0
         DIRECTION  INOUT ; 
         PORT 
             LAYER met2 ;
             RECT     0 1.45 0.5 1.95 ;
         END
    END  in_Y_1_0
    PIN  in_Y_2_0
         DIRECTION  INOUT ; 
         PORT 
             LAYER met2 ;
             RECT     0 2.25 0.5 2.75 ;
         END
    END  in_Y_2_0
    PIN  in_Y_3_0
         DIRECTION  INOUT ; 
         PORT 
             LAYER met2 ;
             RECT     0 3.05 0.5 3.55 ;
         END

--------------------------

lib for XYZ : 

    cell ("XYZ") {
        area : 54.76;
        cell_footprint : "XYZ";
        dont_touch : "true";
        cell_leakage_power : 2.2805530000;
        driver_waveform_fall : "ramp";
        driver_waveform_rise : "ramp";
        pin   ("in_X_0_0") {
            capacitance : 0.0024350000;
            clock : "false";
            direction : "inout";
            fall_capacitance : 0.0023640000;
            max_transition : 500 ;
            rise_capacitance : 0.0025060000;
        }
        pin   ("in_X_1_0") {
            capacitance : 0.0024350000;
            clock : "false";
            direction : "inout";
            fall_capacitance : 0.0023640000;
            max_transition : 500 ;
            rise_capacitance : 0.0025060000;
        }
        pin   ("in_X_2_0") {
            capacitance : 0.0024350000;
            clock : "false";
            direction : "inout";
            fall_capacitance : 0.0023640000;
            max_transition : 500 ;
            rise_capacitance : 0.0025060000;
        }
----------------------------

Any way to send a debug trace ?

vijayank88, 2021-12-20:
@ratul619 
attach `/openlane/designs/xyz/runs/RUN_2021.12.20_10.40.30/logs/placement/5-global.log"`
Also you clk port and in config.tcl why its disabled?

ratul619, 2021-12-20:
i dont have any sequential logic. So disabled the clk port. But if i enable it , during sta run also its failing 

WARNING ORD-1011] LEF master RRAM has no liberty cell.
[WARNING ORD-1011] LEF master sky130_fd_sc_hd__and2_2 has no liberty cell.
[WARNING ORD-1011] LEF master sky130_fd_sc_hd__buf_1 has no liberty cell.
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/sta.tcl |& tee >&@stdout /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_10.59.02/logs/synthesis/2-sta.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

So skilled that , so the run can go on....

ratul619, 2021-12-20:
Pasting the global placement log : 

OpenROAD b79f266fe41540eabc064bcaddfe19ed715ac5c2 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_11.01.25/tmp/merged_unpadded.lef
[WARNING ODB-0220] WARNING (LEFPARS-2036): SOURCE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_11.01.25/tmp/merged_unpadded.lef at line 922.

[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 442 library cells
[INFO ODB-0226] Finished LEF file:  /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_11.01.25/tmp/merged_unpadded.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_11.01.25/tmp/floorplan/4-io.def
[INFO ODB-0128] Design: rram_wrapper
[INFO ODB-0130]     Created 18 pins.
[INFO ODB-0131]     Created 3 components and 45 component-terminals.
[INFO ODB-0133]     Created 19 nets and 21 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/rram_wrapper/runs/RUN_2021.12.20_11.01.25/tmp/floorplan/4-io.def

donn, 2021-12-20:
As Vijayan mentioned, we do have an issue structure you need to adhere to. There are relatively few OpenLane contributors and wildly formatted issues affect our ability to help you.

closed 2021-12-20 by donn