The-OpenROAD-Project/OpenLane #774: Google-Style Copyright Assignment For Contributions

donn, 2021-12-20:
### Prompt
Currently, our copyrights are a bit of a mess. Situations such as #771 come to mind, where people don't know what the proper copyright assignment is.

### Proposal
I find @google's way of doing things uniquely sober: https://opensource.google/docs/releasing/authors/

Essentially, we are to replace the copyright headers with `The OpenLane Authors`. The file already has a link to a list of contributors and a list of major contributors.

@mithro thoughts?

maliberty, 2021-12-20:
The link you provide states "This information is in addition to go/releasing/contributions. It does not replace it.".   The go link requires a CLA which is rather onerous.

donn, 2021-12-20:
That... is true. Gah.

Well I'm at a bit of a loss frankly, as the situation is currently a bit muddy: there are contributors who, despite legally having copyright over their changes, chose to not add their names to headers. So this whole thing is a bit of a gray area.