The-OpenROAD-Project/OpenLane #781: Segmentation violation in sta_multi_corner.tcl

roman3017, 2021-12-21:
### Description
When running user_project_wrapper design I encounter segmentation violation error.

### Environment
```
bash-4.2$ python3 ./env.py issue-survey
Failed to run Docker info.
Python: v3.6.8
Kernel: Linux v5.11.0-43-generic
Distribution: centos 7

Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 2021.12.21_01.44.20
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

commit 4eb363c95978d3480732f0ee3a70b00253a2ca9c
Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>
Date:   Mon Dec 20 18:20:52 2021 -0300

    Add Jenkinsfile for OpenROAD team CI (#778)
    
    [skip ci]
    
    Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>

commit 8c734bc051d0f302abd9a5437688f8dc75ffb32f
Author: Donn <me@donn.website>
Date:   Sat Dec 18 03:32:48 2021 +0200

    Move some layer variables to PDK, update io_place.py (#767)
    
    + `RT_{MIN,MAX}_LAYER` added to the PDK, relies on layer names
    + `FP_IO_{H,V}LAYER` added to the PDK, relies on layer names
    + `DRT_{MIN,MAX}_LAYER` optional override variables added
    ~ Minimum layer set to `met1` in the PDK, you have to enable `li1` manually
    ~ `GLB_RT_LAYER_ADJUSTMENTS` was moved to the PDK
    ~ io_place.py API redone in click, now also uses layer names
    - `FP_IO_{H,V}METAL` removed, will be translated to the new ones on-the-fly if they're in a user's config (for now.)
    - `GLB_RT_{MIN,MAX}LAYER` and its clock equivalents deprecated, will automatically be changed to the new ones on the fly

commit ec570990c76a65b9bf23d89a161dbe075ed8fa15
Author: Donn <me@donn.website>
Date:   Thu Dec 16 19:44:52 2021 +0200

    Remove All Unused Scripts + More Consistent Script Naming (#765)
    
    * Remove Unusued Scripts
    * Rename all `camelCase` scripts to `snake_case` for consistency (except mergeLef.py)
    * More naming consistency, folded useless scripts

bash-4.2$ 
```

### Reproduction Material
* Upload a tarball containing the relevant design.
* List the commands used to run the design.

To reproduce, run:
```
./flow.tcl -design user_project_wrapper
```

### Expected behavior
Succesful temination without segmentation violation.

### Logs
```
./flow.tcl -design user_project_wrapper
...
======================= Slowest Corner =================================

[ERROR]: during executing openroad script /openlane/scripts/openroad/sta_multi_corner.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: segmentation violation

[INFO]: Creating reproducible...
[INFO]: Saving runtime environment...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/user_project_wrapper/runs/RUN_2021.12.21_20.42.52/openroad_issue_reproducible…
⭕️ Done.
[INFO]: Reproducible packaged: Please tarball and upload designs/user_project_wrapper/runs/RUN_2021.12.21_20.42.52/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Calculating Runtime From the Start...
[INFO]: Generating Final Summary Report...
[INFO]: Design Name: user_project_wrapper
Run Directory: /openlane/designs/user_project_wrapper/runs/RUN_2021.12.21_20.42.52
Source not found.
----------------------------------------

LVS Summary:
Source: /openlane/designs/user_project_wrapper/runs/RUN_2021.12.21_20.42.52/logs/finishing/user_project_wrapper.lvs.lef.log
Source not found.
----------------------------------------

Antenna Summary:
No antenna report found.
[INFO]: check full report here: /openlane/designs/user_project_wrapper/runs/RUN_2021.12.21_20.42.52/reports/final_summary_report.csv
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/sta_multi_corner.tcl  -indexed_log $log"
    (procedure "run_sta" line 21)
    invoked from within
"run_sta -log $::env(routing_logs)/parasitics_multi_corner_sta.log -multi_corner"
    (procedure "run_routing" line 83)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 7)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 54)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
        puts_info "Running interactively"
        if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 345)
```

roman3017, 2021-12-21:
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/7758193/openroad_issue_reproducible.tar.gz)

jjcherry56, 2021-12-22:
There are a number of problems with scripts/openroad/sta_multi_corner.tcl.
It reads liberty files for EXTRA_LIBS before calling define_corners, which is not supported.
OpenSTA has to know the corners before any liberty files are read.
Secondly, it makes no attempt to read corner specific  liberty files for EXTRA_LIBS (which is why it seg faulted
looking for the corner data for a ram).

I will make it not seg fault but it has no hope of running the way it is written.

roman3017, 2021-12-23:
Is there a way to disable it then? This check/crash is blocking my progress at the moment and there is just few days left for TO deadline.

jjcherry56, 2021-12-24:
If you remove EXTRA_LIBS you will be missing any paths from/to the sram instances.
There is nothing inherently multi-corner in the analysis; it is really just running three different 
corners at the same time and printing the results separately. 
The simple work around is to rewrite the script into 3 different scripts, one for each corner.

The correct solution is to rewrite the script to read the sram liberty files after define_corners and 
read separate sram liberty files for each corner just like the std cells. This assumes you have liberty
files for each corner for the srams.

If you can't locate the corner liberty files for the srams the following will work to re-use the sram 
at one corner.

```
define_corners ss tt ff
read_liberty -corner ss $::env(LIB_SLOWEST)
read_liberty -corner tt $::env(LIB_SYNTH_COMPLETE)
read_liberty -corner ff $::env(LIB_FASTEST)

foreach corner {ss tt ff} {
  if { [info exists ::env(EXTRA_LIBS) ] } {
    foreach lib $::env(EXTRA_LIBS) {
      read_liberty -corner $corner $lib
    }
  }
}

```
It gives warnings but works correctly,

opensta/openroad both now error if define_corners is called after reading a liberty file rather than seg fault.

roman3017, 2021-12-25:
Thanks, I will try.

PR #822 (2021-12-29): Fix Multicorner Analysis
(Based on comments by James Cherry)

Resolves #781
closed 2021-12-29 by donn