The-OpenROAD-Project/OpenLane #784: Timing Signoff: SS corner seems to be the worst-case corner for hold violations

msaligane, 2021-12-22:
### Description
We have noticed during the timing validation of the [LiteX management core](https://github.com/efabless/caravel_mgmt_soc_litex/releases) that min timing checks has an unexpected trend where `sky130_fd_sc_hd__ss_100C_1v60.lib` seems to be showing the worst case slack. 
Based on my checks of the openlane flow, the following libs are used for multi-corner checks:
```
sky130_fd_sc_hd__ss_100C_1v60.lib
sky130_fd_sc_hd__ff_n40C_1v95.lib
sky130_fd_sc_hd__tt_025C_1v80.lib
```
We have compiled the timing reports in the following table:
[image]

Similarly, as a sanity check,  we have ran the AES design from openlane's design suite regression tests. We noticed the same behavior:

[image]

### Environment
```
---                                                                                                                 Git Log (Last 3 Commits)                                                                                                                                                                                                                commit 96a630e29c60f47d87c63642e2af39c19a48cdb7                                                                     Author: Vitor Bandeira <vvbandeira@users.noreply.github.com>                                                        Date:   Tue Dec 21 21:37:30 2021 -0300                                                                                                                                                                                                      Improve Jenkins pipeline (#782)                                                                                                                                                                                                         - clean before checkout                                                                                             - fix stage names                                                                                                   - add back designs                                                                                                                                                                                                                      Signed-off-by: Vitor Bandeira <vitor.vbandeira@gmail.com>                                                                                                                                                                               [skip ci]                                                                                                                                                                                                                           commit 1ab8d55fab363b9e10b08111a9ee14954f65e3ef                                                                     Author: Donn <me@donn.website>                                                                                      Date:   Wed Dec 22 02:36:50 2021 +0200                                                                                                                                                                                                      Fix or Remove Designs Failing The Extended Test Set (#777)                                                                                                                                                                              + Added ability to run extended test sets on a specific pull request's PR by adding `[ci ets]` anywhere in the PR body (the example there counted!)                                                                                     + Added capacity to disable certain designs in a test set by prefixing the design with a `#`                        + Added new script that allows someone to iterate on a design's timing closure                                      ~ Decreased core utilization for some designs that just aren't routing                                              ~ Fixed sizes of {BM64, blabla, y_huff} to avoid pin location issue                                                 ~ Increased max buffer percentages for some designs failing to achieve timing closure                               ~ Update all references to the design directory in config.tcl for all designs to `$::env(DESIGN_DIR)`.              ~ core area/die area no longer have an EOL at the end of their value                                                                                                                                                                commit 86a5e29c4f9126920174dc2f2e43a1348c743a83                                                                     Author: Donn <me@donn.website>                                                                                      Date:   Tue Dec 21 20:52:09 2021 +0200                                                                                                                                                                                                      Update Documentation for `or_issue.py` (#780)                                                                                                                                                                                           [skip ci]   
```

### Reproduction Material
The design is available in github here (with tar balls and all):  https://github.com/efabless/caravel_mgmt_soc_litex/releases

### Expected behavior
I expect the min timing checks reports to fail for the `ff_n40C_1v95` corner or show worst violating slack compared to the `ss_100C_1v60` corner.

@mkkassem FYI.

maliberty, 2021-12-22:
The first odd thing I see is that the .lib are characterized over different input ranges:

tt:
                cell_fall ("del_1_7_7") {
                    index_1("0.0100000000, 0.0230506000, 0.0531329000, 0.1224740000, 0.2823110000, 0.6507430000, 1.5000000000");
                    index_2("0.0005000000, 0.0013283400, 0.0035289500, 0.0093752600, 0.0249070000, 0.0661697000, 0.1757910000");
ff:
                cell_fall ("del_1_7_7") {
                    index_1("0.0100000000, 0.0230506000, 0.0531329000, 0.1224740000, 0.2823110000, 0.6507430000, 1.5000000000");
                    index_2("0.0005000000, 0.0013908400, 0.0038688600, 0.0107619000, 0.0299362000, 0.0832729000, 0.2316380000");
ss:
                cell_fall ("del_1_7_7") {
                    index_1("0.0100000000, 0.0281727000, 0.0793701000, 0.2236070000, 0.6299610000, 1.7747700000, 5.0000000000");
                    index_2("0.0005000000, 0.0015089000, 0.0045535400, 0.0137416000, 0.0414694000, 0.1251460000, 0.3776650000");

Its not proof of a problem but it seems odd - tt & ff are much closer than ss.

maliberty, 2021-12-22:
Power is even odder - is ss really 1000X leakier?

ff:
    cell ("sky130_fd_sc_hd__a2111o_1") {
        leakage_power () {
            value : 0.0003327000;
            when : "A1&A2&B1&!C1&!D1";
        }
tt:
    cell ("sky130_fd_sc_hd__a2111o_1") {
        leakage_power () {
            value : 0.0017945000;
            when : "!A1&!A2&!B1&!C1&D1";
        }
ss:
    cell ("sky130_fd_sc_hd__a2111o_1") {
        leakage_power () {
            value : 2.6966021000;
            when : "!A1&!A2&!B1&!C1&D1";
        }

msaligane, 2021-12-22:
I think you ll need to open github issues about those. I am also planning to do that regarding the ccsnoise models which are randomly included in specific corners. I am starting to think the reason might be in .libs.

@maliberty Do you think you can run AES in ORFS and check if you see the same behavior?

maliberty, 2021-12-22:
In ORFS for slack:

     min     max
ff -0.15    0.05
tt -0.19   -2.05
ss -0.22   -7.49

msaligane, 2021-12-22:
max timing seems to make sense (also the close timing between ff and tt would be explained by your comment [here](https://github.com/The-OpenROAD-Project/OpenLane/issues/784#issuecomment-999280227). Since you can timing getting worse as you go from ff to ss.

However, min timing seems weird in ORFS as well. Maybe because latency is increasing / skew is increasing in ss, then hold is worse?

maliberty, 2021-12-22:
The hold path is very short and is dominated by clock tree delay.  It looks like both launch and capture branches slow symmetrically on the worst path.  Worst skew does increase ff->tt->ss but not worst hold slack.

donn, 2021-12-22:
@RTimothyEdwards Thoughts?

RTimothyEdwards, 2021-12-23:
You should normally expect worst case setup violations at slow-low-hot and worst case hold violations at high-fast-cold.  If you are seeing hold violations at slow-low-hot that aren't there at typ-typ-typ, there's some chance that you're looking at a false path.  But a hold violation just means that a downstream clock arrived after the transistioning data .  You can get the same hold violation by slowing down the clock to the downstream flop, or by speeding up the transitioning data to the downstream flop.  Since everything is getting faster or slower at a fast or slow corner, then there's no way of knowing which way it will go.  The issues we had with MPW-one had to do with a (effectively) designed-in clock skew, with massive hold violations that had nothing to do with the process corner.  Changes due to corner simulation will be smaller, and more random in appearance.  A setup time difference in 30ns between fast and slow sounds more like an issue with the way the analysis is set up---the entire clock period is 20ns.

tspyrou, 2022-01-05:
@msaligane what do you think. Are you satisfied with this explanation?

msaligane, 2022-01-06:
@RTimothyEdwards Sure, there might be false paths. I have ran now a couple designs in openlane and ss is the worst case (consistently).

My initial theory is that we have a huge latency and skew on the clock tree? which might explain the ss being an issue but we still should see ff trigger MIN violations.

@tspyrou This is existing in ORFS as well so we need to go and check it closely. When the ECO flow is merged and that task is over we can spend cycles on debugging this. 

Finally, a recharacterization of the .libs would definitely help here. But Before that, I will check on other pdks if I see the same issue which might help understand if this is only seen in sky130 (IE. .lib weirdness).

mithro, 2022-01-06:
Here is Mehdi's data presented in a form which makes it easier to understand
[image]

mithro, 2022-01-06:
@RTimothyEdwards is asserting that;
> Changes due to corner simulation will be smaller, and more random in appearance.

A ~30ns change in Max Slack on the LiteX Management Core is not a "small / random in appearance change" category. Something is clearly wrong there -- but there is no evidence if this is being caused by the analysis or a real effect.

Maybe the <4ns change in Min Slack for both the LiteX Management Core and the AES Core does fit the "small / random in appearance" change? The fact that Min Slack on the AES Core isn't linear in one direction like the others does point more to "random".

mithro, 2022-01-06:
However, as @RTimothyEdwards points out -- a ~30ns change if you are using a ~20ns clock period does feel like more likely a mistake rather than something that is real....

tspyrou, 2022-01-12:
@msaligane if the set of meetings convinces you that there is no problem, please close this. If not, please make a testcase with a report_checks path that shows the problem and assign it to me.

msaligane, 2022-01-12:
I think I know what is happening - most of the paths that are failing hold are pin-to-reg and give different results across corners because the sdc (input_delay) isn't updated to slower/faster corner. When I filter down all the reports generated by openlane (group_path: `async_default`). We see that the failing paths are in ff as expected.

So I will close this.

closed 2022-01-12 by msaligane