The-OpenROAD-Project/OpenLane #785: Unable to run "make pdk"

donghl17, 2021-12-22:
### Description
I am not able to run make pdk to update the pdk version because of the following error: 
```
make[1]: Leaving directory `/home/xxx/OpenLane/pdks/skywater-pdk'                                                                                                                                                                        cd /home/xxx/OpenLane/pdks/open_pdks && \                                                                                                                                                                                                        git checkout master && git pull && \                                                                                                                                                                                                         git checkout -qf 476f7428f7f686de51a5164c702629a9b9f2da46                                                                                                                                                                            error: Your local changes to the following files would be overwritten by checkout:                                                                                                                                                                   VERSION                                                                                                                                                                                                                                      sky130/Makefile.in                                                                                                                                                                                                                           sky130/custom/sky130_fd_io/gds/sky130_ef_io.gds                                                                                                                                                                                              sky130/custom/sky130_fd_io/mag/sky130_ef_io__vdda_hvc_pad.mag                                                                                                                                                                        Please, commit your changes or stash them before you can switch branches.                                                                                                                                                                    Aborting                                                                                                                                                                                                                                     make: *** [open_pdks] Error 1
```

### Environment
```
PDK Version Verification Status: MISMATCH                                                                                                                                                                                                    The version of open_pdks installed does not match the one required by the OpenLane flow scripts (installed: 14db32aa8ba330e88632ff3ad2ff52f4f4dae1ad, expected: 476f7428f7f686de51a5164c702629a9b9f2da46)                                    You may want to re-install the PDK by invoking `make pdk`.
```

vijayank88, 2021-12-22:
@donghl17 
Provide full log, OS details, steps you followed

donn, 2021-12-22:
The issue template is mandatory.

closed 2021-12-22 by donn