The-OpenROAD-Project/OpenLane #786: sram macros don't install as expected

d-m-bailey, 2021-12-22:
### Description
`open_pdks`'s `sky130/scripts/configure` allows for the following `--enable-sram-sky130` options

`--enable-sram-sky130="yes"` or `--enable-sram-sky130="YES"` : sets source directory to `../sources/sky130_sram_macros`
`--enable-sram-sky130="no"` or `--enable-sram-sky130="NO"` : sets source directory to `""`
`--enable-sram-sky130=some-directory` : sets source directory to `some-directory`
Note: `--enable-sram-sky130` is processed as `--enable-sram-sky130="yes"`

OpenLane's `Makefile` allows for two `INSTALL_SRAM` settings: `enabled` and anything else.
```
INSTALL_SRAM ?= disabled

OPEN_PDK_ARGS ?= ""
ifeq ($(INSTALL_SRAM), enabled)
OPEN_PDK_ARGS ?= --enable-sram-sky130
endif
``` 
Even if `INSTALL_SRAM` is set to `enabled`, nothing will happen.
`OPEN_PDK_ARGS` is only set to `--enable-sram-sky130` if it was unset, but it was set to "" right before the `ifeq` statement.

The only two ways to install the sram macro library are `export OPEN_PDK_ARGS=--enable-sram-sky130` for default download/install or `export OPEN_PDK_ARGS=--enable-sram-sky130=some-directory` for installation from an existing directory.

### Environment
```
Python: v3.7.3
Kernel: Linux v4.19.0-17-cloud-amd64
Distribution: debian 10
Container Engine: docker v20.10.7
OpenLane Git Version: 2021.11.13_01.43.55

Alert: Click is not installed.
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

commit 6245b0e675d0e6751bdfc5362c545efe29d1077c
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Fri Nov 19 13:38:50 2021 -0600

    Fix sram install (#724)
    
    Also disable by default

commit e0e6fbc1bd90d1c73aae19769d997da8c3c47512
Author: Manar <manarabdelatty@aucegypt.edu>
Date:   Fri Nov 19 12:30:08 2021 -0600

    Add option for enabling/disabling repair_tie_fanout (#722)
    
    * Add option for enabling/disabling repair_tie_fanout
    
    * add escape characters
    
    * Add option for enabling sta report power

commit 92f127435283591857a06477f0ae15d172540d81
Author: Donn <me@donn.website>
Date:   Thu Nov 18 12:06:02 2021 +0200

    Non-fixed layer count for GLB RT adjustments (#721)
    
    * Non-fixed layer count for GLB RT adjustments
    
    Remember this kids, nothing is more permanent than a temporary solution.
    
    * GLB_RT_LX_ADJUSTMENT variables consolidated into GLB_RT_LAYER_ADJUSTMENTS, a comma-delimited set of values
    
    * GLB_RT_LX_ADJUSTMENT will continue to be accepted but display a deprecation warning, and will be removed in a later version of OpenLane.
    
    * GLB_RT_L1_ADJUSTMENT value from the PDK is now ignored.
    
    * Bashrc fix (Line wrap problem), or_issue fix
```

### Reproduction Material
First case:
```
rm -rf pdks
export INSTALL_SRAM=YES
make pdk
```
Second case:
```
rm -rf pdks
export INSTALL_SRAM=enabled
make pdk
```
Third case:
```
rm -rf pdks
export OPEN_PDK_ARGS=--enable-sky130-sram
make pdk
```

### Expected behavior
First case: 
**expectation**: download to default directory and installed 
**result**: not installed

Second case:
**expectation**: download to default directory and installed 
**result**: not installed

Third case:
**expectation**: download to default directory and installed 
**result**: installed

### Logs
First case
from `pdks/open_pdks/scripts/config.log`
```
configure:2598: Package 'sky130_sram_macros' will not be installed.
```
Second case
from `pdks/open_pdks/scripts/config.log`
```
configure:2598: Package 'sky130_sram_macros' will not be installed.
```
Third case
from `pdks/open_pdks/scripts/config.log`
```
configure:2582: Package 'sky130_sram_macros' will be installed automatically during make.
```

d-m-bailey, 2021-12-22:
The code snippet was actually from the `efabless/openlane` fork and has already been partially fixed here.

donn, 2021-12-22:
I'll merge this after mpw-4 is tagged. Thanks, Mitch.

PR #788 (2021-12-22): Install sram enhancement
Adds the flexibility provided by the `open_pdks/scripts/configure` script (handles `yes`, `no` and `directory` correctly).
Resolves #786.
closed 2021-12-22 by donn