The-OpenROAD-Project/OpenLane #789: OpenLane flow get stuck at yosys part "Continuing TECHMAP pass"

ufukyill, 2021-12-22:
Hi 
I was trying to run my verilog code which is about 2k rows through OpenLane in order to get a gds file. 
(Im using ubuntu 18.04 via virtualBox.) When i run the command :  "/.flow.tcl -design top_level" the flow stucks at ,exact same step, "Continuing TECHMAP pass" on every try and the virtual machine does not responds in anyway. Screenshot of log file is attached below. 
(I was able to get a gds file when i ran flow for some modules seperately. Then i put together these modules in a .v file which is compiled and tested).
[image]

vijayank88, 2021-12-22:
@ufukyill 
check how much used for yosys running

ufukyill, 2021-12-22:
After posting i have already increase the RAM of virtual machine and it kept continue. But now i having this error at "Continuing TECHMAP pass" step :
[image]

donn, 2021-12-22:
The issue template is mandatory.

closed 2021-12-22 by donn