The-OpenROAD-Project/OpenLane #793: Error at "Continuing TECHMAP pass" step while while running OpenLane

ufukyill, 2021-12-22:
Hi
I was trying to run my verilog code which is about 2k rows through OpenLane in order to get a gds file. I was able to get a gds file when i ran flow for some modules seperately. Then i put together these modules in a .v file which is compiled and tested. Now it gives this error which is given below as screenshot.
[image]

donn, 2021-12-22:
The issue template is mandatory, @ufukyill. We can't help you if you delete everything and just upload a screenshot.

closed 2021-12-22 by donn

ufukyill, 2021-12-24:
> The issue template is mandatory, @ufukyill. We can't help you if you delete everything and just upload a screenshot.

Which template exactly? Should i choose a bug report or smth? I'm a newbie so do not mind me.

donn, 2021-12-24:
Yes, you should choose "bug report."