The-OpenROAD-Project/OpenLane #796: Allow cell flattening in magic extraction

d-m-bailey, 2021-12-23:
### Prompt
Currently, magic extracts each layout cell separately. This can lead to confusing netlists with unnecessary hierarchies when using parameterized layout cells from GDS. For example, `nmos_m10_w7_000_sli_dli_da_p` in the openram macros.

### Proposal
Allow user specified flattening of cells when reading GDS.
For example, executing these preprocessing commands before GDS input.
```
gds flatglob {*nmos_m*} 
gds flatglob {*pmos_m*}
gds flatglob {sky130*__*\[A-Z\]*} 
gds flatglob {*_example_*}
gds flatglob {*\$\$*}
gds flatglob {*_CDNS_*}
gds flatten yes
```
This could be accomplished by modifying `run_magic_spice_export` of `scripts/tcl_scripts/magic.tcl` as follows
Before:
```
if { \[info exist ::env(MAGIC_EXT_USE_GDS)\] && \$::env(MAGIC_EXT_USE_GDS) } {
        gds read \$::env(CURRENT_GDS)
} else {
```
After:
```
if { \[info exist ::env(MAGIC_EXT_USE_GDS)\] && \$::env(MAGIC_EXT_USE_GDS) } {
       if { \[info exist ::env(MAGIC_GDS_PREPROCESSING_FILE)\] } {
          source \$::env(MAGIC_GDS_PREPROCESSING_FILE)
       }
        gds read \$::env(CURRENT_GDS)
} else {
```
The preprocessing commands would be stored in the file specified by `MAGIC_GDS_PREPROCESSING_FILE`.

### Alternatives
There might be a way to store all the preprocessing commands in a variable and `eval` it.

donn, 2021-12-23:
Ah, god, no eval. This codebase is a security nightmare as is.

But this is a reasonable ask.

d-m-bailey, 2021-12-23:
How about a list of cells to flatten?
Example: 
```
{*nmos_m*} 
{*pmos_m*}
```
Then programmatically loop through the list adding `gds flatten`.
```
if { \[info exist ::env(MAGIC_EXT_USE_GDS)\] && \$::env(MAGIC_EXT_USE_GDS) } {
       if { \[info exist ::env(MAGIC_GDS_FLATTEN_CELLS)\] } {
          foreach cell \$::env(MAGIC_GDS_FLATTEN_CELLS) {
             gds flatten $cell
          }
          gds flatten yes
       }
        gds read \$::env(CURRENT_GDS)
} else {
```