The-OpenROAD-Project/OpenLane #802: Run floorplan error

ufukyill, 2021-12-24:
### Description
I am trying to run the flow of my design interactively on OpenLane. On the synthesis stage my flow does not complete itself so synthesised the design seperately on Yosys and copied the netlist verilog file to run path as runs/run3/results/synthesis/top_level.v.
then as the next step of the flow, I am trying to run floorplan but I get an error like this;

[image]

line 522 of the code: `reg [127:0] outputData;` 
line 523 of the code: `output [127:0] outputData;` 

### Environment
`rb
    scripts = `find ./scripts -type f -maxdepth 1`.chomp.split("\n")
    
    unused_files = []
    
    for script in scripts
        basename = File.basename(script)
        STDERR.puts "Searching for #{basename}..."
        ag_files = `ag --count #{basename}`.chomp.split("\n")
        if ag_files.count > 0
            STDERR.puts "Found #{basename} in #{ag_files.count} files:"
            for file in ag_files
                STDERR.puts "\t-#{file}"
            end
        else
            STDERR.puts "Could not find #{basename} in any files!"
            unused_files << script
        end
    end
    
    STDERR.puts "Unused files: #{unused_files}"
    puts "#{unused_files.map { |f| "'#{f}'" }.join(" ")}"
    
    
    * Rename all `camelCase` scripts to `snake_case` for consistency (except mergeLef.py)
    
    rb
    scripts = `find ./scripts -type f -maxdepth 1`.chomp.split("\n")
    
    unused_files = []
    
    for script in scripts
        if script.downcase != script
            puts script
        end
    end`

maliberty, 2021-12-24:
A reg statement isn't structural verilog so whatever you did in yosys didn't work correctly.

donn, 2021-12-27:
Your issue template is still wrong, but also, what matt said.

closed 2021-12-29 by donn